lvgl/lvgl #6: Why LV_UPSCALE_MAP is removed?

ajaybhargav, 2017-01-09:
Hi I just wanted to know why LV_UPSCALE_MAP is now removed from core?

kisvegabor, 2017-01-09:
Hi!
Because it only makes sense  if antialiassing is enabled and I supposed that with better quality settings  the using of not antialiassed maps are not  effective. 
Nonetheless if you have got an other opinion, pleas tell me!

ajaybhargav, 2017-01-09:
upscaling was very effective with images atleast, to keep images in proportion with screen size. e.g. with 320x240 screen and upscaling enabled I can use an image of 320x240 to fill screen. However with no upscaling I need to create image of double the size 640x480 to fill the screen which takes more memory. upscaling for images saves lot of space and no/minimum loss of quality.

kisvegabor, 2017-01-09:
I see. However I think it could be more effective as an image attribute rather then a global flag. 

And what about the fonts? I removed font upscale first because the upscaled fonts on an antialiassed screen looked ugly.

ajaybhargav, 2017-01-09:
An image attribute will be perfect. I don't see any problem with fonts as of now. I have not tried smaller fonts though, I used fonts of size 30 and 40. 30 looks fine to me and I am sure 40 will look good too.

I would also like to understand and discuss about VDB, to make screen refresh smooth enough. Should I create a different issue for it?

kisvegabor, 2017-01-09:
Ok, I will revert it as an image attribute. So do you agree font upscaling is not required?

VDB: Sure, create a new issue, please!

ajaybhargav, 2017-01-09:
👍 for image attribute.
To be true, since my background image is now in 4 parts coz of no upscaling :) so correlating fonts size with background is difficult. I cannot comment on fonts. But on first look they seems OK to me. Once image upscaling is done I can retest fonts and then figure out if its needed for fonts or not.

kisvegabor, 2017-01-09:
Ok, great!

ajaybhargav, 2017-01-09:
sure let me check it

ajaybhargav, 2017-01-09:
did you push beta branch? as I see it still shows 7 commits behind.

kisvegabor, 2017-01-09:
Uhh, sorry. I pushed now!

ajaybhargav, 2017-01-09:
Scaling works but It works but highly dependent on order in which functions are called. e.g. if I call set_upscale after set_file it does not work coz w and h does not get updated.
maybe this will do better:

```
void lv_img_set_upscale(lv_obj_t * img, bool en)
{
    lv_img_ext_t * ext = lv_obj_get_ext(img);

    ext->upscale = (en == false ? 0 : 1);
    if (ext->upscale != 0) {
        ext->w *=  2;
        ext->h *=  2;
    }
}
```

kisvegabor, 2017-01-09:
Thanks, it is fixed!

kisvegabor, 2017-01-16:
Now it is released in v2.2.0. So I close this this issue!
Thank you! :)

closed 2017-01-16 by kisvegabor