lvgl/lvgl #7: Regarding VDB and display update

ajaybhargav, 2017-01-09:
I dont know how VDB is linked to objects, it seems display area is recreated on VDB everytime display needs to be updated (right?). I have gone through lv_refr.c and found that vdb is filled there and then flushed on to display row by row. Somehow this makes display update slower and one can see line by line update like raster scan. I was thinking it would be great if there is a way VDB gets filled first and then sent to display routine to flush it at once. Smaller VDB size makes it even worse. I am not taking about VDB=0 as its not good at all.

kisvegabor, 2017-01-09:
What is the size od your VDB?
If you have "1 line sized" VDB you can load only 1 line at once. If you have got "half screen sized" VDB the half screen can be loaded at once. The content of the VDB is flushed in lv_refr.c/lv_vdb_flush(). Here the whole VDB is "sent" to the display, but if antialaissing is used the filtering is done here row-by-row.

ajaybhargav, 2017-01-12:
I have big enough VDB but since antialaising is enabled, display is updated row-by-row. Can we do some kind of improvement there?

kisvegabor, 2017-01-12:
It's interesting. Only the filtering of antialaissing is happening row by row but it . Can you send the "Graphical settings" sections of your lv_conf.h?
And what kind of microcontroller do you use?

ajaybhargav, 2017-01-12:
These are the basic setting:
```
#define LV_HOR_RES          (320 * LV_DOWNSCALE)
#define LV_VER_RES          (240 * LV_DOWNSCALE)

/* Buffered rendering: >= LV_DOWNSCALE * LV_HOR_RES or 0 to disable buffering*/

#define LV_VDB_SIZE         (LV_HOR_RES * (LV_VER_RES / 8))

/* Enable antialaiassing
 * If enabled everything will half-sized
 * Use LV_DOWNSCALE to compensate
 * the down scaling effect of antialiassing*/
#define LV_ANTIALIAS        1
/*Set the downscaling value*/
#if LV_ANTIALIAS == 0
#define LV_DOWNSCALE        1
#else
#define LV_DOWNSCALE        2
#endif

#define LV_REFR_PERIOD      40    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average number of objects on a screen */
```
I am using mediatek GSM chipset. It says in spec that frequency can go upto 260Mhz but I really dont feel that much of speed from testing that I have done.

kisvegabor, 2017-01-12:
The lv_conf.h looks good. 
I have a few questions:
1. The refresh happens row-by-row pr pixel-by-pixel?
2. How do you communicate with the display? (SPI, parallel, RGB)
3. Did you try a simple screen fill without the library?

ajaybhargav, 2017-01-12:
I am using ILI9340 display running over SPI. and SPI is fast enough > 10MHz clock. One limitation of SPI driver is I cannot send more than 1024 bytes at once. It takes around 70ms to fill the display. Another thing is I have to swap the bytes of color_t (MSB, LSB) before sending it to display controller. There is so much processing going on before data is actually sent to display.

kisvegabor, 2017-01-12:
An when you are using the library is the refreshing significantly slower then the raw fill? 
Without antialiassing does it become much faster?
A previous question could be important: "The refresh happens row-by-row pr pixel-by-pixel?"

With your settings: `LV_VDB_SIZE         (LV_HOR_RES * (LV_VER_RES / 8))`
the refresh should heppen in 8 steps. If there is no big delay between the steps then it looks the problem is in a lower level. May be 260 MHz is less as you mentioned.

ajaybhargav, 2017-01-12:
Actually as of now to make things faster, I made few changes:
1. Added a new function to flush changes to display at once. (inside hal/disp/disp.c)
```
void disp_flush(uint8_t disp_id, int32_t x1, int32_t y1, int32_t x2, int32_t y2)
{
#if USE_ILI9340
    ili9340_flush(x1, y1, x2, y2);
#endif
}
```
2. There is a local buffer also added of size equal to size of display, which gets filled during disp_map and flushed to display during disp_flush.
```
//Map only does this much
    int32_t i;
    uint16_t act_w = act_x2 - act_x1 + 1;
    uint16_t act_h = act_y2 - act_y1 + 1;

#if COLOR_DEPTH == 16
    for (i = 0; i < act_w * act_h; i++) {
    	*pfb++ = (color_p[i].full >> 8) & 0xFF; /* swapping of bytes */
    	*pfb++ = color_p[i].full & 0xFF;
    }
#else
```
I also feel swapping of bytes eat good amount of time. Fill is very fast coz buffer once prepared is flushed repeatedly.
3. Call to disp_flush is like this (lv_vdb_flush()):
```
		disp_area(DISP_ID_ALL, vdb.vdb_area.x1 >> 1, y, vdb.vdb_area.x2 >> 1, y);
		disp_map(DISP_ID_ALL, row_buf);
	}
	disp_flush(DISP_ID_ALL, vdb.vdb_area.x1 >> 1, vdb.vdb_area.y1 >> 1, vdb.vdb_area.x2 >> 1, vdb.vdb_area.y2 >> 1);
```

This is how I am currently speeding up things.

kisvegabor, 2017-01-12:
Can you send a video about a screen refresh with the library?

ajaybhargav, 2017-01-13:
Take a look:
https://goo.gl/photos/xm3F8ZAFWygVoXVR9

kisvegabor, 2017-01-13:
Thank you!
From the video it can be seen clearly: the display is refreshed in 8 steps because your VDB size is 1/8 screen. It seems the MCU can fill the VDB slowly. Keep in mind with antialiassing your display resolution looks doubled (pixels x4).

kisvegabor, 2017-01-13:
If you can use screen sized VDB then the screen would be refreshed in 1 step.

ajaybhargav, 2017-01-13:
yes MCU does not work as fast as it is mentioned in the document. That's a different story.
The screen update you see is done in one shot and 8 steps only after I added those changes to VDB (display flush thing). else row by row is even slow. So I was thinking if we can improve it somehow. Coz right now I am doing double buffering one in VDB for row construction and other in display driver for screen construction. Its hard to leave the benefits of antialiasing :)

kisvegabor, 2017-01-13:
Ok, I have an idea to not refresh row by row without extra buffer! :) I can do it to Monday!

Your SPI sending is nonblocking? So do you have to wait transfer or it goes from interrupt.

ajaybhargav, 2017-01-13:
That will be awesome, Let me know when to test. Its blocking transfer I don't have a control on that since there is no source provided just APIs, that makes it even worse :)

PS: I added more patches to symbol font pull request. Review it.

kisvegabor, 2017-01-16:
I released the v.2.2.0 version which contains the update. Now the antialiassed data is written back into VDB. So it can be flushed as one large data.

kisvegabor, 2017-08-18:
Already a some updates arrived related to VDB:
* Make the anti-aliasing back to VDB
* Optimize anti-aliasing
* Double VDB support (great with SPI) read more [here](http://www.gl.littlev.hu/overview#drawing)

I suppose they solve your problems with VDB. 
Do you agree if I close this issue?

ajaybhargav, 2017-08-21:
I do not have a setup ready for testing so for now, we can close this issue. I will reopen in case I find anything. Thanks!

closed 2017-08-21 by ajaybhargav