lvgl/lvgl #8: New object Gauge

ajaybhargav, 2017-01-09:
Is it possible to add a gauge object, similar operation to that or chart/progress bar maybe. Gauges looks great on displays and make it look more intuitive.

kisvegabor, 2017-01-09:
I was thinking about it. The only problem is its needle, because I can draw only a line. To draw a great needle  image rotation would be needed.
However if you think a simple line is enough as needle then it can be added.

ajaybhargav, 2017-01-09:
A simple line would do great I guess. It's not bad to start with, later can be modified to make needle using a polygon maybe? something like this...
[image]

kisvegabor, 2017-01-10:
I added an initial version of Gauge to 'beta'. (Misc also has to be updated to the 'beta'.)
You can test it with the attached code. Any feedback is welcome.
[gauge.zip](https://github.com/littlevgl/lvgl/files/697235/gauge.zip)

ajaybhargav, 2017-01-13:
I just tested gauge, It looks great... I think it will look much better if we move the datum of needle to base of gauge scale so that when gauge is set to 0, needle looks flat (hope you understand what I am trying to say :) ).  Or better make it as an option, so that gauge can also be used for making analog clock dials.

Just a thought, Is it possible to have option to make it semi circle instead of a complete one?

kisvegabor, 2017-01-13:
Thanks for the review! The angle of the gauge can be set in its style. If you set it to 180 then it will be flat. 

To make it semi circle you should put it on a transparent object with half gauge height and align the gauge properly. But I can add an option for that.

ajaybhargav, 2017-01-13:
great 👍 are you going to work on needle shape?  I am not too good in this stuff else I could help you somehow :) But I can do all beta testing.

kisvegabor, 2017-01-16:
I released an initial version of Gauge in v.2.2.0. The mention updates might be implemented in a next release. Thanks for the idea, and the tests! :)

kisvegabor, 2017-12-21:
Gauge is reworked in v5.0.0 to look more realistic. You can check it [here](http://www.gl.littlev.hu/object-types/gauge-lv_gauge)

closed 2017-12-21 by kisvegabor

ajaybhargav, 2017-12-21:
looks great.. 😄 lvgl has improved a lot. I am yet to test the newest version. I am not getting chance to be online on GitHub.

kisvegabor, 2017-12-21:
Thank you :)
I'm waiting for your feedback when you will have some free time! :)