lvgl/lvgl #11: ide compability

ramonschepers, 2017-05-27:
i am desperate to get this to the Arduino IDE, but i get a huge list of errors when compiling and i can't figure it out due to lack of knowing lvgl well enough... perhaps port it to arduino as well?

kisvegabor, 2017-05-27:
Have you added the project's root path to the include paths?

Can you copy here a few from the error messages?

ramonschepers, 2017-05-27:
well, arduino is set up differently, and i don't know how to do that on arduino ide.
so i copied everything to the sketch folder, with adding the needed includes in the sketch file.

then i came across alot of "file not found" (due to root include paths), fixed those manually, but then i got "undefined reference to" errors, so i had to include every .c file in my sketch too, and now i get even more errors saying this:

C:\Users\Admin\AppData\Local\Temp\arduino_build_78852\sketch\lvgl/lv_objx/lv_bar.c: In function 'lv_obj_t* lv_bar_create(lv_obj_t*, lv_obj_t*)':

C:\Users\Admin\AppData\Local\Temp\arduino_build_78852\sketch\lvgl/lv_objx/lv_bar.c:62:42: error: invalid conversion from 'void*' to 'lv_bar_ext_t*' [-fpermissive]

     lv_bar_ext_t * ext = lv_obj_alloc_ext(new_bar, sizeof(lv_bar_ext_t));

why not simply make everything easy to integrate in arduino too? :P

ramonschepers, 2017-05-27:
note that this is with the application framework enabled

kisvegabor, 2017-05-27:
Hm. Please attache the whole compile log as a txt file.

ramonschepers, 2017-05-27:
okay, i set compiler warnings to "all" so beware of a HUGE list of errors, here's the list: https://pastebin.com/dJZmD2rp

kisvegabor, 2017-05-27:
It seems like compiiler threat the source files as cpp files instead of c. Have you changes the file extensions to .cpp from .c?

ramonschepers, 2017-05-27:
the arduino ide does that by default i think, and i cant find a way on how to disable that, and i leaved the file extentions at their default ones ( .h, .c )

kisvegabor, 2017-05-27:
I googled on it and found that it a limitation of the Arduino IDE. I have never used Arduino so far but others say there are problems when you begin to use things outside of Arduino. Therefore they recommend to work with an other IDE. 

https://stackoverflow.com/questions/7803416/can-the-arduino-ide-be-used-to-develop-plain-old-c-programs

Here is a description about a C compatible IDE:
https://spin.atomicobject.com/2011/10/15/taking-the-training-wheels-off-the-arduino/

ramonschepers, 2017-05-27:
i see, so that basically means i need to port the entire device support package to another IDE... (the teensy 3.x support package in my case, which will be a pain)

and i was so hopeful that lvgl could work relatively easy on arduino.. :'(

ramonschepers, 2017-05-27:
just a question, will it ever be ported to arduino? or not?

kisvegabor, 2017-05-27:
Yes, I'm planning to port it to Arduino (and make the required updates) but I will have time for it only a few month  later... :(

Teensy 3 doesn't support other IDE/compiler then Arduino?

ramonschepers, 2017-05-27:
awesome that it's going to be ported! no worries on time, do everything at your tempo, will donate some money for the awesome support :)

and for the teensy 3 not as far as i know.. ^^;

ajaybhargav, 2017-05-27:
You can create a wrapper class around lvgl to make it arduino compatible, converting whole project to c++ will be a big job.

ramonschepers, 2017-05-27:
never did that kind of coding before, so i wouldn't know how to do that tbh ^^;

kisvegabor, 2017-05-27:
I don't know Arduino, but I suppose that regardless to the wrappers the Arduino compiler has to compile the code. The wrappers helps you only when you use the functions, don't they?

ajaybhargav, 2017-05-27:
I have been working on Arduino recently to add support for a new platform that I am making. And lot of code that I write is in C. Arduino core is actually mix bag of C and C++ sources so I am not sure why compiler is throwing errors. It may be the way you are including the files in your project/sketch.

kisvegabor, 2017-05-28:
I've downloaded the Arduino IDE and tried to setup an example dummy library. The first thing I faced that I had to modify the .h files: 
at the begginig
`#ifdef __cplusplus`
`extern "C" {`
`#endif`

at the end
`#ifdef __cplusplus`
`} // extern "C"`
`#endif`

If you have time for that can you try it?

Also to not waste time I recommend you to try the PC simulator. It is a very good to know the LittevGL and also your code can be copied to an embedded environment later. Here is a guide about the PC simulator: http://www.gl.littlev.hu/pc-simulator

ramonschepers, 2017-05-28:
you mean at every .h file? that's gonna take some time.

and i will try the pc emulator :)

ramonschepers, 2017-05-28:
also, regarding the pc simulator, why not make a single visual studio project? if needed i can give you a project that's exactly doing that :)

kisvegabor, 2017-05-28:
Yes, I mean all .h files... :(

Visual studio project: I've chosen Eclipse CDT because it is cross platform. please create an [issue in proj_pc](https://github.com/littlevgl/proj_pc/issues) repository to continue the discussion there.

ramonschepers, 2017-05-28:
Okay, will change all files and try again upcoming week and tell you the result

ramonschepers, 2017-06-01:
i midified all the .h files, and now getting "undefined reference to lv_init()" and the same errors with other functions.

kisvegabor, 2017-06-02:
Thank you for the effort to modify the files.

Please, send me the lvgl, misc and hal folders and I will check/try them.

ramonschepers, 2017-06-02:
attached is the arduino folder including ili9341 and touch libraries for the adafruit 2.8" tft ( https://www.adafruit.com/product/1651 ) with a teensy 3.1 (i used teensyduino 1.36 & arduino 1.8.1)

the whole zip needs to be extracted in my documents/Arduino
[Arduino.zip](https://github.com/littlevgl/lvgl/files/1046996/Arduino.zip)

kisvegabor, 2017-06-02:
I've checked the files, tried to compile and saw multiple type of issues. On next week I'll try to solve them!

kisvegabor, 2017-06-08:
Firstly I have tried to port the misc library and I've succeeded! I needed to solve several issues because of the limitations in the Arduino IDE but I was able to compile it! Please try it and if it is working for
 you too I will make the LittlevGL as well. 

[Arduino_misc.zip](https://github.com/littlevgl/lvgl/files/1061246/Arduino_misc.zip)

ramonschepers, 2017-06-08:
it compiles, but it doesn't go past the misc_init(); function on a teensy 3.2 (fastest, with lto, 96mhz), aka a system halt is thrown.

kisvegabor, 2017-06-08:
In misc_conf.h disable the IDLE module. USE_IDLE 0

ramonschepers, 2017-06-08:
ok, it goes past the misc_init() now, but the ram usage didn't increase, as i see this on the serial monitor:
```
Start
Free RAM:44035
Misc init.. Done!
Free RAM:44035

```

kisvegabor, 2017-06-08:
It's good if the RAM usage is not growing after init. All the variables are already allocated regardless to calling `misc_init`. Try not to `#include <misc.h>` and not call `misc_init()`. It should decrease the RAM usage.

Other question: Have you ever tried this Eclipse based IDE for Arduino? http://www.baeyens.it/eclipse/
To be hones I'm not so happy with the current solution because (just to mention one thing) I have to use relative include paths which makes the project messy.

ramonschepers, 2017-06-08:
never tried that IDE, so would have to try first.

ramonschepers, 2017-06-08:
well, this is funny, i tried to use a display driver with the hal files (disp.c + disp.h)
i got multiple errors, and now i am trying to write a bare bone driver in the disp files themselves.

still lots of errors, seems like littlevgl doesn't like cpp class usage in the code? (like spisettings class to adjust spi sleep, mode and such with one call)

this is quite dramatic on my side right now ^^;

kisvegabor, 2017-06-09:
The lvgl is still not modified like misc it won't work with Arduino now.

Please, give a try to suggested the Eclipse IDE. It seems very difficult to handle such a  complex project like a GUI in Arduino IDE.

kisvegabor, 2017-06-09:
Here is guide about how to add the Teensy board to this Eclipse IDE: 
https://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/arduino-eclipse-ide-2-4-arduino-1-6-3-teensyduino-1-22-mac-windows-installation/

kisvegabor, 2017-06-24:
Hi! Have you tried the Eclipse baded IDE? If yes what are your experiences?

ramonschepers, 2017-06-24:
Still have to try, been quite busy the past period

ramonschepers, 2017-07-13:
Sorry, I can't find the motivation to move to a different IDE, seems I am too hooked on the Arduino IDE.. ^^;

Maaaybe will try to in the future.
I'm leaving it up to you to close the issue, or simply post updates regarding Arduino IDE support on this thread.
Offtopic: right now I am having the urge to make my own GUI library for the Arduino IDE again.

kisvegabor, 2017-07-14:
Sad to hear this. Unfortunately the Arduino IDE compatibility could mean a step back for me because the lack of project properties.

If later you decide to use this library I can offer the already mentioned  ([Sloeber](http://eclipse.baeyens.it/index.shtml))  Eclipse IDE.

Anyway, thank you for the contribution so far!

ramonschepers, 2017-09-25:
i installed the sloebel ide, imported the teensy core, and get errors in the teensy core with spiclass (which i need for using the SPI bus.)

this is really frustrating..

kisvegabor, 2017-09-27:
Hi,

It is good to here that you have tried that IDE.

Were you able to solve the issue meanwhile?

ramonschepers, 2017-09-27:
Sadly not :(

kisvegabor, 2017-11-26:
Hi Ramon,

Good news for Arduino! In the upcoming v5.0 release I modified the architecture the graphics library and now it is compatible with the default Arduino IDE. I was able to easily compile it with Teensy 3.6.

I cannot test it because I don't have Teensy hardware, but I hope you can do the testing! :) You need to write a *disp_fill* and *disp_map* function (see below)

Here is the library: 
[lvgl_arduino_lib.zip](https://github.com/littlevgl/lvgl/files/1504289/lvgl_arduino_lib.zip)

An this is my sketch:
```c
#include <lvgl.h>

void hal_init(void)
{
    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    disp_drv.fill = NULL;  /*function to fill an area:          void fill(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t color) */
    disp_drv.map = NULL;   /*function copy pixels to an area:   void map(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t *color_p) */
    disp_drv.copy = NULL;  /*ignored*/
    lv_disp_register(&disp_drv);

    /*Initialize the graphics library's tick*/
    /*In a Timer call lv_tich_handler() in every milliseconds*/
}

void setup() {
  lv_init();

  hal_init();

  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino!");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

ramonschepers, 2017-11-26:
awesome, altho i see a problem, the arduino IDE has a macro called map, which is used in LV_HAL_DISP.H

ramonschepers, 2017-11-26:
hmm... i get errors when assigning my own function to the null pointers:
```
error: expected primary-expression before 'x1'
     disp_drv.map = disp_map(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t *color_p);
```

kisvegabor, 2017-11-26:
I updated the library to not use 'map' name. 
[lvgl_ardion_lib2.zip](https://github.com/littlevgl/lvgl/files/1504346/lvgl_ardion_lib2.zip)

```c
    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    disp_drv.fill_fp = NULL;  /*function to fill an area:          void fill(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t color) */
    disp_drv.map_fp = NULL;   /*function copy pixels to an area:   void map(int32_t x1, int32_t y1, int32_t x2, int32_t y2, lv_color_t *color_p) */
    disp_drv.blend_fp = NULL;  /*ignored*/
    lv_disp_register(&disp_drv);
```

You get the error because you have to initialize it like this:`disp_drv.map_fp = disp_map` So write onyl the function name.

ramonschepers, 2017-11-27:
ok, didn't know that

now i get this error(?)
error: 'lv_disp_drv_t {aka struct _disp_drv_t}' has no member named 'fill'

kisvegabor, 2017-11-27:
Look the previous code example! To be consistent with `map_fp` I renamed `fill` to `fill_fp`

ramonschepers, 2017-11-27:
yeah, i figured it out, i am pretty sleepy right now

got it to compile, going to test soon

ramonschepers, 2017-11-27:
it compiled, but doesn't work with my ili9341, while it does work when i call the tft driver library (the non lvgl) functions directly, so it must be something inside of lvgl that refuses to run.

btw, i am running a tick timer for the lvgl as documented

ramonschepers, 2017-11-29:
Just thinking out loud now: the teensy support package runs as multi thread within each yield call (which happens alot!)
This might be a problem with the lvgl scheduler to prevent it from executing?
Also, the MCU doesn't freeze up, I can draw in the loop what and whenever I want.

Also note that the lvgl doesn't just give no output on the screen, but also no change on the SPI bus to which the screen is connected to.

Yes I have configured the lvgl config file for correct screen resolution, also reduced buffer to 10 lines to fit on my teensy 3.2

kisvegabor, 2017-11-29:
Oh sorry, I forget to add `lv_task_handler()` to the loop... You can also add a  few millisecond delay after it and/or call your applicaton's functions.

ramonschepers, 2017-11-29:
it works, lvgl clears the screen to black, but.... no text is displayed :P

kisvegabor, 2017-11-29:
Please show me your disp_map function.

ramonschepers, 2017-11-29:
```

void disp_map(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_p) {
  uint16_t c;
  tft.setAddrWindow(x1,y1,x2,y2);
  for (int y = y1; y <= y2; y++) {
    for (int x = x1; x <= x2; x++) {
      c = tft.color565(color_p->red, color_p->green, color_p->blue);
      tft.drawPixel(x,y,c);
    }
  }
}
```

kisvegabor, 2017-11-29:
It seems you forget to increment `color_p` :)

ramonschepers, 2017-11-29:
this proves how newb i am with lvgl xD
thanks! it works as should, not i can dive into optimizing my disp_map function, as it is kinda slow.

issue can be closed :)

kisvegabor, 2017-11-29:
Awesome ^^

Please keep in mind the `dev-5.0` is under development and some changes may still happen until it is released.

closed 2017-11-29 by kisvegabor

Pablo2048, 2017-11-29:
@ramonschepers Instead of .drawPixel try to use .pushColor - this may speed things up... ;-)