lvgl/lvgl #12: Could you share the STMproject source?

qwert1213131, 2017-06-09:


kisvegabor, 2017-06-09:
Do you mean the project or the source code of the GUI?

qwert1213131, 2017-06-10:
it is about the code ported to stm32.
i have the stm32f429 disco board.
i can't make the gui worked on stm32.
thanks!

kisvegabor, 2017-06-12:
I've added a download button for the project tothe bottom of this page: 
http://www.gl.littlev.hu/blog/8/embedded-gui-on-stm32-discovery-board-with-littlev-graphicslibrary

qwert1213131, 2017-06-13:
Thank you very much！

kisvegabor, 2017-06-24:
Had you time to try it? And was it working?

Tiweee, 2017-06-26:
Hi, I tried that on the same STM board, but it is lagging a lot. Where could be the problem? Thank you

kisvegabor, 2017-06-26:
I've updated the project. Please, check it again.

Tiweee, 2017-06-27:
Thx now it´s working perfectly. Maybe i just did something wrong because i use IAR ide so i had to convert whole project. Please do you have some demo project or code of aplications (like sys. monitor) in use? I cant make them to work.
Thank you and your library is awesome. Great work! 👍

kisvegabor, 2017-06-27:
Your welcome! Good to hear that!

I don't have a different project for the applications because it is very easy to make them work. Actually in the project you are using the applications are enabled and initalized just the demo creates an other screen so the applications are not visible. 

You just need to comment out the `stm32_test_init();` in `main.c` to see the applications. So far only a few applications are exists however I constantly create new ones. 

Please, check the bottom section in `lv_conf.h` where you can find the application related configurations.

If you are interested in the applications please start an other thread for them. Now, because the demo project seems working for you too, if you agree I close this issue.

closed 2017-07-07 by kisvegabor