lvgl/lvgl #13: Can I get an offline version of the help documentation?     pdf？

qwert1213131, 2017-07-10:
I’m learning the source code，and have ported to the stm32412g-disco.
That's amazing.  
The screen is 1.54 inches, a little, suitable for wearable design.
[image]

kisvegabor, 2017-07-10:
I'm happy that it is working for you! 

I'm not planning to write an offline documentation, because I suppused the online (html) is more useable.
However it seems you think it differently. Why do you think the offline doc. is better?

qwert1213131, 2017-07-11:
The site is slow to open .
Is there a plan to add an introduction to the animation section? 
Thank you!

kisvegabor, 2017-07-11:
Take a look at this: http://www.gl.littlev.hu/misc-library#anim
Does it help?

Writing and maintaning an offline documentation too is a lot of work. But if it will be important and useful I add it to the TODO list.

qwert1213131, 2017-07-11:
Thank you very much！

qwert1213131, 2017-07-11:
I've run animation successfully, but some frames are missing.
[http://v.youku.com/v_show/id_XMjg4NDQ0MjY0OA==.html?spm=a2h0k.8191407.0.0&from=s1.8-1-1.2](http://v.youku.com/v_show/id_XMjg4NDQ0MjY0OA==.html?spm=a2h0k.8191407.0.0&from=s1.8-1-1.2)

kisvegabor, 2017-07-11:
It seems to me no frames are missing only the rendering is slower. You can increase the LV_VDB_SIZE in lv_conf.h.

Do you use anti-aliasing? (lv_conf.h LV_ANIALIAS). If yes then you should check with NO anti-aliasing.

kisvegabor, 2017-07-11:
Please, start a new thread for the animations directly.

qwert1213131, 2017-07-11:
OK,I will have a try! thanks!

kisvegabor, 2017-08-09:
I'm thinking about the offline documentation but it still seems a lot of work the create and maintain. It would be a better deal if people can download it for a few dollars (e.g. $10). Would you pay this small amount of money for an offline documentation?

kisvegabor, 2018-03-09:
I was thinking about the offline documentation and I found a workaround. There are tools to generate PDF from HTML sites. For example this worked fine for me: https://www.sejda.com/html-to-pdf
I hope it also fulfill your needs.

As now I'm not planning to simultaneously maintain an offline documentation too I close this issue. 

Thank your for idea! :)

closed 2018-03-09 by kisvegabor