lvgl/lvgl #14: Using input widgets without a toutchscreen

PTDreamer, 2017-07-15:
Hi,

I would like to use the library on a system with just a clickable rotary encoder as an input device. How can I accomplish that since the library doesn't seem to have the active widget state? 

i.e. rotate the encoder changes the active button (or list item or whatever) and clicking on it sends a "click" to the active widget.

kisvegabor, 2017-07-15:
Hi,

Yes, you are right. There is no "native" support for this feature. But I like it! Thank you for the idea.

I've already figured out how to realize it. On Monday I will send you a example implementation.

kisvegabor, 2017-07-17:
What do you think about it?
https://www.youtube.com/watch?v=1CVRU0p1JGI&feature=youtu.be

[enc_main.zip](https://github.com/littlevgl/lvgl/files/1152398/enc_main.zip)

PTDreamer, 2017-07-17:
Yeh that is exactly what I need, lists and comboboxes require auto scroling to the active item though

PTDreamer, 2017-07-17:
Ok I have looked at the attached code and understand your solution. I was thinking about the possibility of making the active state a property of the graphical object with it's own style (e.g. LV_STYLE_BTN_ACTIVE). Also having the library handling the active states automagicaly by having functions to change the active item to the next or previous item (like tab indexes on frameworks like Qt or .NET), and to send signal to them like activeitemclick()

kisvegabor, 2017-07-17:
Great!

Yes was thinking something similar too. This functionality can be integrated to the library but it mean a lot of work to implement all the handler functions of different object types.

kisvegabor, 2017-07-19:
I had some free time so I implemented one part of it. Now you can create groups from objects and activate next or previous. A callback function belongs to every object group to modify the style of the active object. For example: `color_mix(style->mcolor, COLOR_ORANGE, OPA_50)` will mix orange to the current color therefore it is not required to define new styles. 

The next step is to implement new signals when an object is activated and its value increased/decreased. 

https://www.youtube.com/watch?v=FUDmUKaejFg&feature=youtu.be

[enc2.zip](https://github.com/littlevgl/lvgl/files/1159189/enc2.zip)

What do you think?

qwert1213131, 2017-07-27:
There are many devices not equipped with touch screens, so it is necessary to use buttons。
That's great

kisvegabor, 2017-07-28:
I added a full featured sub system called **lv_group** to the `beta` branch. 

See an example here: https://github.com/littlevgl/lvgl/blob/beta/lv_examples/2_2_encoder_ctrl/encoder_ctrl.c
In the example there are buttons on the GUI to simulate encoder actions. 

Can you try it?

kisvegabor, 2017-08-18:
The update is released in v4.2.0.
Read more here: http://www.gl.littlev.hu/objects/#group

closed 2017-08-18 by kisvegabor

PTDreamer, 2017-08-19:
I'm sorry for the lack of feedback, I'm currently on vacations and don't have the hw required to test the new code.
Thank you for the new features.

kisvegabor, 2017-08-19:
No problem, just enjoy your holiday! :)
I'm waiting for your feedback when your hardware will be ready.