lvgl/lvgl #15: Support for non-blocking display interface functions

iamniels, 2017-08-04:
First of all, great lib!

On most microcontrollers, copying the data from the memory to the display can be handled by a DMA controller or timers. If such method is used, the CPU is not utilized during data transfer to the display. So the CPU can perform other tasks, like rendering the next part of the screen.

In virtual drawing mode this can speedup the GUI a lot. Instead of one buffer, use two buffers. While the first buffer is being transferred to the screen by the DMA controller, the second buffer is filled by the CPU. 

What do you think? 
What needs to be changed in the lib to support this? At least the blocking interface API functions need to be changed.

kisvegabor, 2017-08-04:
Hi,

thank you!

Very good and relevant idea. It seems it can be implemented without modifying the current API. Only some extensions are required.

I try to add these feature in the next days. Can you help in testing is?

iamniels, 2017-08-04:
First I need to get the basic version running on my hardware. Then I will test it.

kisvegabor, 2017-08-08:
I've implemented it on my STM32F429 Discovery board and measured the render time of a quite complex GUI with different settings. Here are the results:

GPU |VDB [lines]|Double VDB| Time [ms]
------ | ------------| -------------| -------------
   0   |       20     |       0       |   243
   1   |       20     |       0       |   237
   1   |       20     |       1       |   243
   0   |       80     |       0       |   153
   0   |       80     |       1       |   157
   1   |       80     |       1       |   151
   1   |      160    |       0       |   137

Surprisingly only the VDB size have affected the render time.

**Why?**
* GPU: have a big advantage only with a lot of opacity else could be bad because of configuration overhead
* Double VDB: Lets assume copying one byte from VDB to frame buffer  takes e.g. 4 instructions (simple byte copy) Then copying 320 x 240 x 2 bytes (16 bit colors) at 180 MHz: 
`153.600 byte x 4 inst /180.000.000 = 3 ms` which is not so much. In addition DMA interrupts takes some time too.

**Could it be useful?**
Yes, for example if you are using external display controller and transfer pixels via SPI or parallel port which might be much slower. In this case the flushing could last even 50 ms.

**How to use the double VDB?**
* Enable it `lv_conf.h` with `#define LV_VDB_DOUBLE     1` 
* In `disp.c` in `disp_map()` transfer pixels with DMA (or as you want) and call `lv_vdb_flush_ready()` when the transfer is ready. 
* Please, use `beta` branch until this feature is not released. 

Thank you for the good idea again and I'm waiting for your feedback if you have also tried it!

iamniels, 2017-08-08:
Thanks, great!

I will test it and let you know what I think.

iamniels, 2017-08-17:
I've tested it with [my display driver](https://github.com/iamniels/stm32_ili9341_display_driver). I have not tested it extensively because I just started with LVGL. But it seems to run fine. Thanks!

kisvegabor, 2017-08-18:
I'm glad to hear that it is working there. 
Was it easy to integrate the LittlevGL or were there some confusing parts?

iamniels, 2017-08-18:
It was easy to integrate. The split between hal, lvgl and misc is nice. One thing I ran into was that the hal contains `disp_init`, but it is not used. 

Also I would like to see documentation about the inner working of LittlevGL, this would make it much easier to contribute. Also the implementation of `color_t` needs more documentation. I correctly guessed I could use color_t.full to get the 16-bit value.

kisvegabor, 2017-08-18:
Yes, `disp_init()` is optional. I supposed it logical to have function like this to make good abstraction of your display.

I got similar feedback from others about a deeper documentation. True, it is nearly impossible to 
 seriously contribute without it. I will work on that!

Have you found this about colors? http://www.gl.littlev.hu/misc-library#color
If not I will add more links to it in the online documentation.

iamniels, 2017-08-18:
I missed that part about the colors in the documentation.

I see you have merged double VDB into master, so I will close this issue.

closed 2017-08-18 by iamniels

kisvegabor, 2017-08-18:
I will add more links to misc library documentation.

Thanks you again for the great idea of *double VDB*!