lvgl/lvgl #16: 3D render options?

ramonschepers, 2017-08-11:
Is there any chance of 3D functions to the engine? 
Like lights, polygons, meshes, vertexes.
(There are some repos on GitHub for MCU's that have some of these features)

kisvegabor, 2017-08-11:
Thank you for the idea!
* Can you mention some cases very it can be useful on a GUI?
* And can you offer some of those repos?

ramonschepers, 2017-09-22:
Sorry for my late comment, 

it would be usefull for minigames, but also ui elements, like pebble did with their smartwatches

A small gif
https://developer.pebble.com/assets/images/tutorials/advanced/action-completed.gif

Will look up some repos up right now

ramonschepers, 2017-09-22:
Here's one: https://github.com/Nand-e/Tiny3D-engine-STM32

kisvegabor, 2017-09-23:
Thank you. It seems resource friendly. But unfortunately it is written in C++. :(

> it would be usefull for minigames

Yes, it would be eye-catching to see a 3D demo game. But I'm not sure it worth the effort. In addition LittlevGL is GUI library, it rather looks a brand new project.

> A small gif
https://developer.pebble.com/assets/images/tutorials/advanced/action-completed.gif

I think it is a gif (or more images played) and not a 3D rendered model.

So in my opinion 3D like elements in an embedded GUI library could appear as an animation of pre rendered images. What do you think?

ramonschepers, 2017-09-25:
To be precise: it is not a sees of images on the Pebble hardware, those animations are rendered real time, with drawing paths, to let em look like 3D and animated like a gif.

So, yeah, since 3D won't be a standard feature, maybe make a extension to lvgl? Like a separate repo which is optional to have?

kisvegabor, 2017-09-27:
The seperated repo sounds better! A compatible 3D engine would be nice to play and to try/learn 3D.

Probably in v6 it could be in the library. (For v5 already has a lot of new things)

Do you have any experience in 3D field and would you like to contribute in this feature?

ramonschepers, 2017-09-27:
I sadly don't have experience with 3D rendering at all, so I can't help much at all

tevfik, 2018-03-11:
I have some codes with my custom Gui.  It needs polygon drawing functions. 3d calculations are made with quaternion and/or Euler arithmetic  (rotation matrices). It requires drawing object pixel size buffer. (may be it's better to use line buffers or sprites but I couldn't handle this yet.). You need to import 3d model as vertices and colors. But could be modified for 2d graphics. Library doesn't have z-buffering, lighting, shading and texturing feature, only rotation, scaling and flat shading is available now. I don't have much time but anyone want to contribute I could help. 

I used this routines with Funkos' modified fooeyGUI framework. So It could run even on a sensor node (like ti's sensortag with a sharp memory display). that gui running as a thread with contiki operating system.

tevfik, 2018-03-11:
By the way library complately optimized with fixed point math library don't use any floating point numbers so it's fast enough If I have time I would upload an video soon...

kisvegabor, 2018-03-12:
@tevfik  It sounds very well. However LittlevGL is not a 3D library an optional 3D extension can be useful in some cases. 
Maybe you already know LittelvGL has a buffered mode where we have a pixel buffer. 
You said there is no z-buffering now.  Without it can you render models properly with overlapping triangles?.

I'm really looking forward to see the video :)

By the way we are talking about polygon drawing in this topic: #146

ramonschepers, 2018-03-15:
[image]
just a small thing i saw in the airplane last night in their entertainment systems. the plane and globe were both a 3D model, and you could do simple navigation around the plane and globe. this might be a optional, but useful thing.

kisvegabor, 2018-03-16:
Wow, it look great! Thank you!
 Once we will have 3D render extension  creating a demo like this would be awesome.

kisvegabor, 2018-07-15:
I'm organizing the issues and however, it's an interesting one, I close it because adding a 3D engine is a very big work and here is not plans to add it in the near future. 

Despite of it if you or somebody make some progress or experiment in this field feel free to comment here and reopen the issue.

barbiani, 2020-03-04:
Hello everybody,

I have made some progress in this topic today. Got [TinyGL](https://bellard.org/TinyGL/) running GLgears smoothly on the ESP32 (M5stack).

[image]

kisvegabor, 2020-03-04:
Wow, that's amazing! Congratulations!

Would you mind sharing your code? We have `lv_lib_...` repos for 3rd parts libs. TinyGL could be added similarly if you are open to it.

barbiani, 2020-03-04:
I am not finding the libraries repository. Also, i am not able to reopen this issue to discuss possible optimizations to this implementation.

embeddedt, 2020-03-04:
> I am not finding the libraries repository.

We would create a new external library repository. [Here](https://github.com/littlevgl?q=lv_lib&type=&language=) is the current list.

barbiani, 2020-03-06:
> Wow, that's amazing! Congratulations!
> 
> Would you mind sharing your code? We have `lv_lib_...` repos for 3rd parts libs. TinyGL could be added similarly if you are open to it.

The solution is quite simple. Let TinyGL render to a canvas buffer... but I am having trouble getting the colors right with 16 and 32 bit colors.

First attempt had 2 buffers for GL and the canvas copying the frame after it was rendered.

The second rendered directly to the canvas buffer and invalidating it after.

While lvgl is 32: ARGB8888, TinyGL is RGBA8888. This troubles SDL... TinyGL has a RGB_TO_PIXEL(r, g, b) macro to handle the pixel format, but the alpha byte is not used. I set it to FF manually.

TinyGL rendering to 16: RGB565 fits the embedded systems, but the canvas does not seem to suport it yet.

kisvegabor, 2020-03-06:
> The solution is quite simple. Let TinyGL render to a canvas buffer... but I am having trouble getting the colors right with 16 and 32 bit colors.

I'm not familiar with TinyGL so don't how difficult to make it work. 

> TinyGL rendering to 16: RGB565 fits the embedded systems, but the canvas does not seem to suport it yet.

The canvas should support RGB565 mode if `LV_COLOR_DEPTH` is `16` in `lv_conf.h`.

Could you add a link to your project? I could take look to see what can be the problem.
Besides, I'm very curious and would like to try it :slightly_smiling_face:

barbiani, 2020-03-06:
> I'm not familiar with TinyGL so don't how difficult to make it work.
>
>
> The canvas should support RGB565 mode if `LV_COLOR_DEPTH` is `16` in `lv_conf.h`.
>
> Could you add a link to your project? I [...]

[Here](https://drive.google.com/file/d/1R_0NzxRh4itaUQPGoZJKmccsYgD1wu6l/view?usp=sharing)

embeddedt, 2020-03-07:
Quickly copied over the necessary files and tested it on STM32. It appears to be running quite smoothly although I don't have time to check the frame rate tonight. :+1:

kisvegabor, 2020-03-07:
Thank you! I've tried it in the simulator and it works well here too.

@barbiani Do you know if there is a way to load model and its textures to TinyGL and set camera position?

barbiani, 2020-03-07:
> Thank you! I've tried it in the simulator and it works well here too.

Do expect wrong colors in the rendering!

> 
> @barbiani Do you know if there is a way to load model and its textures to TinyGL and set camera position?

I would start here https://en.m.wikibooks.org/wiki/OpenGL_Programming/Modern_OpenGL_Tutorial_Load_OBJ

TinyGL can render at least triangle lists, light sources, textures and camera positioning. Says its readme.

embeddedt, 2020-03-07:
I was able to make the last loader @barbiani sent work, with some modifications.

You need this [OBJ file](http://kixor.net/dev/objloader/test.obj), this [MTL file](http://kixor.net/dev/objloader/test.mtl), and the code I've attached below. You can then replace the call to `gears_update` with `showmodel_update` and change `gears_init` to `showmodel_init (CANVAS_WIDTH, CANVAS_HEIGHT, cbuf, "test.obj")`.

At this point the viewport gets filled with color, so I assume I'm looking at a 3D model. However, I must admit that I have no idea how to move the camera with TinyGL, so I don't know whether this is actually being drawn properly or not.

```c

/*
 * showmodel.c -- obj model loader (modified for TinyGL+LittlevGL)
 *
 * Copyright (c) 2005-2007 David HENRY
 * Modifications made by embeddedt
 *
 * Permission is hereby granted, free of charge, to any person
 * obtaining a copy of this software and associated documentation
 * files (the "Software"), to deal in the Software without
 * restriction, including without limitation the rights to use,
 * copy, modify, merge, publish, distribute, sublicense, and/or
 * sell copies of the Software, and to permit persons to whom the
 * Software is furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be
 * included in all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 * EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
 * MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 * NONINFRINGEMENT.
 * IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR
 * ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF
 * CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION
 * WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */

#include <stdio.h>
#include <stdlib.h>
#include "tinygl/gl.h"
#include "tinygl/zbuffer.h"
#include "tinygl/gears.h"

ZBuffer* frameBuffer;

/* Vectors */
typedef float vec3_t[3];
typedef float vec4_t[4];

/* Vertex */
struct obj_vertex_t
{
  vec4_t xyzw;
};

/* Texture coordinates */
struct obj_texCoord_t
{
  vec3_t uvw;
};

/* Normal vector */
struct obj_normal_t
{
  vec3_t ijk;
};

/* Polygon */
struct obj_face_t
{
  GLenum type;        /* primitive type */
[... 533 lines cut ...]
void
showmodel_update (void)
{
  glClear (GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);
  glLoadIdentity ();

  glTranslatef (0.0f, 0.0f, -15.0f);

  /* Draw the model */
  RenderOBJModel (&objfile);

  //glutSwapBuffers ();
}

void
showmodel_init (int w, int h, void *cbuf, char *filename)
{
    frameBuffer = ZB_open(w, h, ZB_MODE_5R6G5B, 0, NULL, NULL, cbuf);
	// initialize GL:
	glInit(frameBuffer);
	glClearColor(0.0, 0.0, 0.0, 0.0);
	glViewport(0, 0, w, h);
	init(filename);
}
```

barbiani, 2020-03-08:
This is also my first opengl attempt.

Looks like setting the camera is in gluPerspective (45.0, w/(GLdouble)h, 0.1, 1000.0)

TinyGL is an opengl subset, so the latter documentation can be used.

https://www.khronos.org/registry/OpenGL-Refpages/gl2.1/xhtml/gluPerspective.xml

Your post has:
/* Initialize OpenGL context */
  glClearColor (0.5f, 0.5f, 0.5f, 1.0f);

Tried myselft to get a grey background but this rgb at .5 turned out yellow. Something was not right. This is probably the color you are seeing.

kisvegabor, 2020-03-08:
I've found the root of the color issue. There was a bug in `RGB_TO_PIXEL` macro (zbuffer.h). Here is the corrected version:
```c
/* 16 bit mode */
#define RGB_TO_PIXEL(r,g,b) \
  (((r) & 0xF800) | (((g >> 5)) & 0x07E0) | ((b) >> 11))
```

@embeddedt 
That's great that you find this example! I couldn't figure out which version of OpenGL TinyGL is compatible with.    
I've tried your code but it crashed for me. The problem is that it finds 7 normals, but still says "has normal: no" therefore the array of normals is not allocated and `NULL` is referenced.
```
first pass results: found
   * 17 vertices
   * 3 texture coords.
   * 7 normal vectors
   * 12 faces
   * has texture coords.: no
   * has normals: no
```

I've created the [lv_lib_tinygl](https://github.com/littlevgl/lv_lib_tinygl) repo for easier code sharing.
@barbiani, I've granted write access to you.
@embeddedt, could you upload your code there?

barbiani, 2020-03-08:
> I've found the root of the color issue. There was a bug in `RGB_TO_PIXEL` macro (zbuffer.h). Here is the corrected version:
> 
> ```c
> /* 16 bit mode */
> #define RGB_TO_PIXEL(r,g,b) \
>   (((r) & 0xF800) | (((g >> 5)) & 0x07E0) | ((b) >> 11))
> ```

I tried editing initScene() to make red, green and blue colors pure. The array format seems to be RGBA. Do that and you will see the color errors I was talking about.

`void initScene() {
	static GLfloat pos[4] = { 5.0, 5.0, 10.0, 0.0 };
	static GLfloat red[4] = { 1.0, 0.0, 0.0, 1.0 };
	static GLfloat green[4] = { 0.0, 1.0, 0.0, 1.0 };
	static GLfloat blue[4] = { 0.0, 0.0, 1.0, 1.0 };
...
}`

Correct look:
[image]

kisvegabor, 2020-03-08:
I see the issue. However, if I set `glClearColor(0.0, 0.0, 1.0, 1.0);` or `glClearColor(0.0, 1.0, 0.0, 1.0);` in `gear_init` I got the correct colors in the bacground. Probably It's an other bug in the renderer.

kisvegabor, 2020-03-08:
I've found it. It was really a bug deep in the renderer. Actually it was a critical thing: `GLParam` is an union with many types. An op_list (an array of `GLParam`) is built from it where opcodes and values are stored. However, at some point, the array was converted to `float *`. I suppose not `float` is the largest member of the union so indexing the array this way made the parameters corrupted. It was easy to fix, but I wonder how many of similar issues could be there.

Now it works for me with 16 bit color depth.

I uploaded my current version to `lv_lib_tinygl`.

barbiani, 2020-03-08:
You are fast! I did not know it was a bug there because I managed to get the right gear colors, but  not the grey background by changing the RGB_TO_PIXEL macro and the gear color definitions.

The original tinygl package has more files than I had in my project.

One of them is LIMITATIONS:

`
Here are listed the functions that TinyGL understands with the known
limitations. The non mentionned functions are *not* implemented and
must not be used.

************ glEnable / glDisable

- The following flags are handled:

GL_CULL_FACE, GL_LIGHTING, GL_COLOR_MATERIAL, GL_TEXTURE_2D, GL_NORMALIZE,
GL_LIGHTx, GL_POLYGON_OFFSET_FILL, GL_POLYGON_OFFSET_POINT,
GL_POLYGON_OFFSET_LINE

- GL_DEPTH_TEST is accepted, but it is only used for lines. For all
  other primitives, Z buffer use is assumed.  The DepthMode is
  hardcoded as GL_LESS (OpenGL default).

************ glShadeModel

OK.

************ glCullFace

OK.

************ glPolygonMode

OK.

************ glBegin

No tests are performed to prevent some functions of being executed
between glBegin/glEnd.

************ glEnd

OK.

************ glVertex

Some prototypes are not implemented.

************ glColor

Some prototypes are not implemented.

************ glNormal

Some prototypes are not implemented.

************ glTexCoord

- Some prototypes are not implemented.

- The Matrix transformation is not applied yet.

************ glEdgeFlag

OK. The edge handling has to be tested, although it is not much useful
in TinyGL.

************ glMatrixMode / glLoadMatrixf / glLoadIdentity / glMultMatrixf /
             glPushMatrix / glPopMatrix / glRotatef / glTranslatef / glScalef /
             glFrustum

- The prototypes with doubles are not implemented.

************ glViewport

GlViewport calls a function pointers to tell glx (or another display
system) to resize the Z buffer and the ximage. Made optional in
version 0.2.

************ glGenLists / glIsList / glNewList / glEndList / glCallList

OK.

************ glClear / glClearColor / glClearDepth

The whole zbuffer and the colors are cleared in any case. The clear color
can be redefined, by *not* the initial z value.

************ glRenderMode

Only the modes GL_RENDER and GL_SELECT are implemented.

************ glSelectBuffer / glInitNames / glPushName / glPopName / glLoadName

OK.

************ glGenTextures / glDeleteTextures / glBindTexture 

OK. These functions should be used to get the maximum performance with
TinyGL. 

************ glTexImage2D

The function accepts only RGB UNSIGNED_BYTES bitmaps. They are
internally resized to 256x256 so you'd better use that size. No
mipmapping is implemented although it will come if asked. No borders
are implemented.

************ glTexEnvi

The only supported mode is GL_DECAL, although others are planned if
asked.

************ glTexParameteri

The other prototypes are not implemented. Only the follwing mode are
implemented:

  glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MIN_FILTER, GL_NEAREST);
  glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MAG_FILTER, GL_NEAREST);
  glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_WRAP_S, GL_REPEAT);
  glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_WRAP_T, GL_REPEAT);

************ glPixelStorei

The pixels are alware byte aligned.

************ glMaterialfv / glMaterialf / glColorMaterial

OK.

************ glLightfv / glLightf / glLightModeli / glLightModelfv

OK. The OpenGL lightening is implemented but not optimized.

************ glFlush

Ignored.

************ glHint

Ignored.

************ glGetIntegerv

- only GL_VIEWPORT, GL_MAX_MODELVIEW_STACK_DEPTH, 
GL_MAX_PROJECTION_STACK_DEPTH, GL_MAX_LIGHTS, GL_MAX_TEXTURE_SIZE,
GL_MAX_TEXTURE_STACK_DEPTH

************ glGetIntegerv

- only GL_TEXTURE_MATRIX, GL_PROJECTION_MATRIX, GL_MODELVIEW_MATRIX,
GL_LINE_WIDTH, GL_LINE_WIDTH_RANGE, GL_POINT_SIZE, GL_POINT_SIZE_RANGE

************ glPolygonOffset

- only API implemented.

************ glEnableClientState, glDisableClientState,
             
- No support for GL_INDEX_ARRAY_POINTER or GL_EDGE_FLAG_ARRAY_POINTER

************ glVertexPointer, glNormalPointer, 
             glColorPointer, glTexureCoordPointer

- OK

------------------------------------------------------------------------------

TinyGL GLX emulation:

************ glXQueryExtension

Returns always True

************ glXChooseVisual

Only 8 bit Pseudocolor or 16 bit Truecolor Visual are accepted. The
attribute list is ignored.

************ glXCreateContext

The sharing is not implemented although the code could handle it.

************ glXDestroyContext

OK.

************ glXMakeCurrent

Not all the syntax is supported yet, in particular with the 'NULL' or
'None' parameters.

************ glXSwapBuffers

OK.

************ glXWaitGL / glXWaitX

Ignored.

See README.BEOS for BeOS limitations.
`

barbiani, 2020-03-08:
For the 3d model rendering I believe that we should parse the obj file, removing comments, including the mtl and changing any texture bitmap to the supported format and writing to a const byte array.

kisvegabor, 2020-03-10:
Let's wait until @embeddedt will have time to upload his code to `lv_lib_tinygl`.

barbiani, 2020-03-12:
Further research found this. Might give good results with lvgl

https://gitlab.com/drummyfish/small3dlib

embeddedt, 2020-03-12:
Looks nice! It definitely appears to have been optimized for memory-constrained environments.

We could probably even get it to render directly through LittlevGL's rendering system instead of needing a separate buffer+canvas.

kisvegabor, 2020-03-13:
Really an interesting option, although I'm concerned about the rendering quality if small3dlib due to integer arithmetics.

IMO if the user needs 3D he can afford the use of float for better quality. What do you think?

embeddedt, 2020-03-15:
I played with `small3dlib` a bit today and it would definitely be a good fit for someone trying to recreate a retro game on a microcontroller. I'm not sure how useful it would be for rendering "modern" 3D models, but for simple games it might suffice.

Performance may be an issue - I ran it rendering directly to the framebuffer (i.e. without LittlevGL) and my STM32F746 was only able to get 1-2 FPS at 480x272 resolution. I did not have the CPU cache on, however. That probably would have given much better results.

Going back to TinyGL, my code appears to be in `objtest.c`. It does look like some modifications have been made to it though. In particular, `ReadOBJModel` should not be commented out [here](https://github.com/littlevgl/lv_lib_tinygl/blob/4fb9d8e0582e27b8a8f1d7aad80d73f491ebf05a/src/objtest.c#L569-L570).

embeddedt, 2020-03-15:
> If the user needs 3D he can afford the use of float for better quality. What do you think?

I agree. The author of `small3dlib` actually writes [here](https://forum.freegamedev.net/viewtopic.php?f=22&t=11747#p85621) that calculations with HW floating-point operations would be faster than fixed-point integer operations in some cases.

embeddedt, 2020-03-15:
I'm also wondering whether full-on 3D rendering capabilities are necessary. Consider the 3D transformation features available in CSS nowadays. If we had the ability to draw things on the canvas and rotate them in 3D space, the plane information interface linked [above](https://github.com/littlevgl/lvgl/issues/16#issuecomment-373516211) could be recreated without the need to render any 3D models.

[image]

* Instead of a 3D globe model, a series of 2D images could be used, or a flat 2D map could be used.
* Perhaps the compass underneath the plane could rotate instead of the plane itself. Then the plane could be a static 2D image.
* The rest of the overlays could be drawn as regular text and images rotated in 3D space.

kisvegabor, 2020-03-17:
> Performance may be an issue - I ran it rendering directly to the framebuffer (i.e. without LittlevGL) and my STM32F746 was only able to get 1-2 FPS at 480x272 resolution. I did not have the CPU cache on, however. That probably would have given much better results.

Am I remember well that you have seen a good FPS with TinyGL?

>  Consider the 3D transformation features available in CSS nowadays. 

I'm learning texture mapping another basics of 3D rendering to see how to realize a "simple" perspective effect on a canvas. Something like [rotate-y in CSS](https://developer.mozilla.org/en-US/docs/Web/CSS/transform-function/rotateY). 

> Instead of a 3D globe model, a series of 2D images could be used, or a flat 2D map could be used.

What about the plane itself? It also could be a 3D model which is rotated. I've seen car dashboards where the model of the car was rotated, moved, zoomed and the doors opened and closed.

> Perhaps the compass underneath the plane could rotate instead of the plane itself. Then the plane could be a static 2D image.

With rotate and perspective effect it really could be a normal image.

---

I'll try the updated code again. I'm still busy with `dev-7.0` and probably will have time for it only at the end of the week.

embeddedt, 2020-03-17:
>  you have seen a good FPS with TinyGL?

Yes. It was much smoother.

kisvegabor, 2020-03-17:
Then it seems like a clear decision here. If TinyGL seems faster and has better quality then we should go with it.

kisvegabor, 2020-03-20:
@embeddedt I've added a [lv_tinygl_test.c](https://github.com/littlevgl/lv_lib_tinygl/blob/master/lv_tinygl_test.c) which uses the mentioned model but it still crashes for me due to the same inconsistency:
```
   * 17 vertices
   * 3 texture coords.
   * 7 normal vectors
   * 12 faces
   * has texture coords.: no   #How? There are 3 as stated above
   * has normals: no           #How? There are 7 as stated above
```

Could you take a look is it already worked for you? Only `lv_tinygl_test()` needs to be called. 

---

Maybe I missed it but where the MTL file is loaded?

embeddedt, 2020-03-20:
@kisvegabor I think I've fixed that issue now, but I am still unable to get anything to appear on the screen besides the gray background.

kisvegabor, 2020-03-25:
Thank you for the fix. It works better but I also see only the gray background. I've tried to understand what's going on inside TinyGL but find nothing so far. It's new territory for me so there are a lot of unclear parts.

kisvegabor, 2020-04-17:
I didn't have time to figure out why we can't see any output and now I'm working with full capacity on v7. :disappointed: 

I still belive the 3D extension would be very useful, but I have no time deal with it now.

kisvegabor, 2020-05-09:
I close this issue in lack of activity :frowning_face: 
Feel free to comment here if you are interested in working on it.

closed 2020-05-09 by kisvegabor

Satnet, 2024-06-05:
> I close this issue in lack of activity ☹️ Feel free to comment here if you are interested in working on it.

I lack the knowledge to push this forward. But there really should be 3D support in LVGL. It would be a great addition!

qq963598371, 2024-06-10:
maybe  https://blog.kaaass.net/archives/1340 is useful

agentdavo, 2024-10-03:
A few years ago I started to port TinyGLES to a fixed point format math.
For MCUs without floating point hardware and to support as far as OpenGL ES 1.1

Based on TinyGLES (c) 2014 Ryan Hileman
Based on TinyGL 0.4 (c) 1997-2002 Fabrice Bellard.

Someone here may be able to finish what I started.

https://github.com/agentdavo/tinygles

agentdavo, 2024-10-13:
GL4ES on top of TinyGLES might be an option
https://github.com/ptitSeb/gl4es

colesnicov, 2024-12-28:
So man. What is the option?

agentdavo, 2024-12-28:
I started to make a CNC controller front end.

My module “cncvis” in this project uses TinyGL reworked by cchads. There are other TinyGLs  but I’m not sure which is best.

https://github.com/agentdavo/uCNC-machineSimModule/

If you compile that you should be able to walk around WS,AD,QE using LVGL event handles and also move the example robot arms 6 joints by holding down 1 thru 6 and arrow keys up/down.

agentdavo, 2025-06-25:
If anyone is interested, I have gears spinning on a canvas using the lv_port_pc_vscode (SDL2)

https://github.com/agentdavo/lv_port_pc_vscode