lvgl/lvgl #18: Suggestion: Move application out of lvgl

ajaybhargav, 2017-08-18:
It is a small suggestion to keep library clean and independent. Is it possible to move lv_app/lv_appx out of lvgl source as they are not part of library as such. May it can be moved to a new repo just like hal/misc etc. This way lvgl as a source will have no code that may or may not be usable to others. One can always close the lvapp repo if he/she needs.

kisvegabor, 2017-08-18:
Do you mean `lv_app` and `lv_appx` too? Just because `lv_app` contains more general things like keyboard or notices.

I'm open to any idea to reorganize the application level. Even the operation principle of application to make them more usable. 

I think the base concept is good to create something which makes a complete, complex task instead of you. However that part clearly needs some good idea! :)

zhaozhongchen, 2017-08-19:
Hi:
first of all, lvgl is a very good lib.
But i think we should let lvgl lib independent, so we can build this as library file (e.g. liblvgl.a on stm32 or liblvgl.so on linux).

1. Remove all app code from lvgl, add merge misc code if we really need.
2. Not all boards has touch screen, but gpio key is chip, on linux, mouse and keyboard as easy to handler.
so we need create a input class to deal with all those input event.
3, I know you create a simulation project for linux, and that project need SDL work. On embeded linux, usually SDL is unavaliable, there should has a module draw image buffer to framebuffer.

kisvegabor, 2017-08-19:
Hi!

Thank you for writing!
 
I don't know what is the advantage of using library files instead of source code. I just see the updating of some library files on releases always needs time and effort. But please correct me!

1. Some modules of `misc` is highly used by `lvgl`. But maybe a minimal configuration of misc is reasonable.
2. Please take a look at  this: http://www.gl.littlev.hu/objects/#group It is an interface to use the library without touchpad, only with buttons or keyboard. Native mouse support is clearly missing right now.
3. Display drivers are not part of the graphics library. They can be implemented via `hal` to keep the library hardware independent. I use the [hw repo](https://github.com/littlevgl/hw/tree/master/per/psp/pc) to add drivers independently. I can imagine to add Linux frame buffer drivers here. What do you think?

My initial goal was to realize your project with just starting and connecting applications. But it doesn't seem viable. So they need to be reworked to become more usable. But why move them to an other repo? Why is it not enough if you can disable them with a define?

zhaozhongchen, 2017-08-21:
Hi:
It's a good news that new object *group* can supports hardware keyboard, but is there has an example in proj_pc?

Yes, some modules of misc is highly used by lvgl, why not merge them to lvgl lib? keey misc independent unreasonable.

Most of app code in lvgl is not suit for other project, others need another UI,  so we just need keey the lvgl lib clean. But we should keey that common code, such as lv_app_notice. In my mind, proj_pc is a good place for that APP code.

ajaybhargav, 2017-08-21:
as @zhaozhongchen also suggested, some part of misc can be merged as they are tightly coupled with lib. So lib becomes independent of hardware as well as application code. This gives an advantage of having lvgl compiled as a static library or shared object as mentioned. This will make lvgl more clean and usable. Of course current "app" code is a good starting point for anyone who is going to make UI or testing lvgl. For some app might be directly usable But for most they going to make their own UI.

ajaybhargav, 2017-08-21:
@kisvegabor I will be happy to help in case you need me for code reorganization.

kisvegabor, 2017-08-21:
**Adding misc**
For example `misc/gfx` is clearly GUI related and can be moved to lvgl. But what to do with *fifo*, *linked list*, *ptask* (simple scheduler), *ramfs* which also used by lvgl? 
With separation of misc and lvgl my goal was to develop a general purpose library along with the graphics library. Now misc contains tested and maintained code which easily can be used in non-GUI related projects too.
Although I'm not an expert with  .so or .a files but I think it is possible generate a lib file from a project containing well configured misc and lvgl code too. Is it true? Because then we can keep the advantages of the separation.
  
**Applications**
Now lv_app has a quite complex background system. But from now lets consider apps only examples for a task and remove the complex background system. Hence they become cleaner and easier to use or modify. Do you agree?
But what to do with built-in notices or keyboard. Shall they integrated to the core library are just remove them? 

**Non-touch control**
@zhaozhongchen you are right if the library supports keyboard usage then it should provide and example code too. I will add it soon. 
Do you think that kind of system (lv_group to focusing and controlling objects) usable? Or do you have any ideas to add?

ajaybhargav, 2017-08-21:
> For example misc/gfx is clearly GUI related and can be moved to lvgl. But what to do with fifo, linked list, ptask (simple scheduler), ramfs which also used by lvgl?

Not all misc need to be merged. My suggestion would be to have misc as a sub-module to this repo instead of explicitly fetching it as a separate repo altogether. In that way misc will still be maintained separately out of lvgl but stable/tested version can always be fetched alongside lvgl library as a submodule and will still act as a part of lvgl. So noone has to bother even if misc development is going on lvgl will always checkout the submodule commit tested/released with lvgl.

> Now lv_app has a quite complex background system. But from now lets consider apps only examples for a task and remove the complex background system. Hence they become cleaner and easier to use or modify. Do you agree?
Disabling LV_APP_ENABLE will disable the whole app and its related components, clearly they are not needed when it comes to lvgl library. So it will be much easier to separate out parts that deal with application only. 

> But what to do with built-in notices or keyboard. Shall they integrated to the core library are just remove them?
I have not gone through the code as such but if keyboard is linked to application the probably input HAL needs improvements so as to enable use of keyboard/touch based on underlying hardware. a HAL property something like get_input_type() -> returning touch or keypad might be useful to enable keyboard interface or touch for navigating and overall GUI interaction.

kisvegabor, 2017-08-21:
**Misc**
You are right now it is inconvenient to keep track lvgl and misc too. Add it as a sub-module seems reasonable and easy to do! In this case we still should keep *misc_conf.h* and *lv_conf.h* separately, shouldn't we?

**Applications**
Do you agree with simplifying apps to stand alone GUI examples too?

**Input devices**
Keyboard means a GUI keyboard like on smart phones. However yes, lvgl can provide better support to real keyboard or mouse too. Support via `hal` sounds great!
_Mouse_ It's like touchpad but a mouse icon should be drawn.
_Keyboard_ Different story, it has to use the focusing mechanism. But the keys can be get automatically and sent to the focused object.

ajaybhargav, 2017-08-21:
I know its a lot to ask but this can be done slowly and steadily. That's the reason I marked it as a suggestion moving ahead.

> In this case we still should keep misc_conf.h and lv_conf.h separately, shouldn't we?

Keeping misc_conf.h separate will not harm. It will give a better control and not clutter lv_conf.h

> Do you agree with simplifying apps to stand alone GUI examples too?
If app is moved out of lvgl then you can not just one but many apps as an example. Even complex apps which shows the power of lvgl. Lets say in a single app folder you have say:
"Touchkeyboard_app_example"
"Tabs_app_example"
"Buttongroup_app_example"
So app as a separate can become a showcase + get started guide for users.

For input devices I still feel hal is very weak, If you remember we discussed about hal having option to register drivers. I think there is lot which can be improved in hal making even hal as part of library just like misc and hardware code completely independent of lvgl. so hardware drivers can register to hal and lvgl only ask hal for properly of hardware it is running on.

kisvegabor, 2017-08-21:
> I know its a lot to ask but this can be done slowly and steadily. 
Probably it's my weakness but how to move misc to lvgl steadily? 
All those things can happen only in a major release (v5.0). 

> So app as a separate can become a showcase + get started guide for users.
I like it! @zhaozhongchen do you agree?

**Add hal to lvgl**
Sounds great too! It will make the library more complete and unified.
Registering drivers can happen with initializing structures with function pointers or other data.

ajaybhargav, 2017-08-21:
>Probably it's my weakness but how to move misc to lvgl steadily?

Create a development branch and start there... all things we do in dev branch will be marked for milestone for v5.0 release.

> Registering drivers can happen with initializing structures with function pointers or other data.

Exactly, this way library when compiled as a static or shared object will expose HAL api for any kind of hardware to register itself to library.

kisvegabor, 2017-08-21:
> Create a development branch and start there... all things we do in dev branch will be marked for milestone for v5.0 release.

Still I'm planning v4.3 with new object types but maybe they will come to v5.0 depending on the progress. 

As I see there are 3 tasks:
1. add misc as sub-module: seems not a big deal just rename a lot of includes in lvgl.
2. update hal: as we discussed with registering drivers 
3. turn application to examples in new repo

ajaybhargav, 2017-08-21:
> Still I'm planning v4.3 with new object types but maybe they will come to v5.0 depending on the progress.

New objects can wait for some time... I think moving misc and hal in lvgl is primary thing to do. then taking out app. I can help you with hal and misc let me know if you want to break the tasks into smaller ones and assign something to me.

zhaozhongchen, 2017-08-21:
It's great that the architecture is getting clearer!

> Registering drivers can happen with initializing structures with function pointers or other data.
Yes, the lvgl lib create some APIs to register all kinds of drivers, then register them when hw init. 

> Mouse It's like touchpad but a mouse icon should be drawn.
we should create an object for mouse cursor, make sure this object always in the top layer.

kisvegabor, 2017-08-21:
Awesome!

@ajaybhargav  it would be great if you could manage the `hal` and `misc` integration.  I can improve the core library to handle mouse and keyboard besides touchpad. As *systick* is also part of  *hal* it should be handled too. Please start a new issue for this thread and discuss the details there.

I can deal with taking out applications and move them to an lv_example repo. I will move the current examples there too.

I will handle the new object types as a smaller priority but I would like to add them to v5.0 for marketing purposes. New "design" elements in the new major release would be engaging.

ajaybhargav, 2017-08-21:
> Please start a new issue for this thread and discuss the details there.

It will sound odd but I do not know what to write in the issue :) Should I just mention "Add hal and misc to lvgl"?

kisvegabor, 2017-08-21:
Right it doesn't sounds like an issue. Do you have a better idea? Talk on the pull request's comment sections?

ajaybhargav, 2017-08-21:
I think I will just create a pull request and then you can review it.

ajaybhargav, 2017-08-23:
can you also create dev-5.0 in lvgl?

kisvegabor, 2017-08-23:
`dev-5.0` branch is created.

kisvegabor, 2017-08-25:
I've started to integrate the new `hal` and `misc` to `lvgl` but I faced a problem. Now the order of the initialization is:
1. Register display driver
2. Init. _lvgl_

However now _lvgl_ initializes _misc_ which is needed by the _hal_  therefore display driver registration fails. 
Besides with the integration of  _misc_ to _lvgl_  general low level drivers can use _misc_ because the include path is not known: *misc/path/to* or  *lvgl/misc/*path/to*

Currently I haven't got any good idea to keep _misc_ in _lvgl_ and ensure the opportunity of usage of _misc_ independently from _lvgl_.

kisvegabor, 2017-09-12:
Hi,
is there any news in this topic?
Now it seems to me misc should be out of lvgl because it is required by other modules and its initialization is not clear.
However hal integratins looks good!

ajaybhargav, 2017-09-13:
Hi Gabor,
I am extremely tied up with my office work right now so couldn't give enough time to hal. I agree we can keep misc out of lvgl for now till the time we figure out a better solution to the problem.

Do you have any suggestions for HAL? I might get some time in coming weekend so I will look into it.

kisvegabor, 2017-09-13:
No problem, there is no deadline. Just deal with it when you have FREE time.
Actually I hasn't time to work on it as well. 

Then I will move misc back, next to lvgl.

`systick` is the last part of hal and is is quite simple. Only `systick_get` is required from the user and `systcik_add_cb`, `systick_rem_cb` optionally. `systick_elapsed` can be implemented by us, because it uses only systick_get.

As have some time I will modify the current `indev` driver to work with the new hal.

kisvegabor, 2017-12-21:
Hi,

thank you for the contribution! 
Today the new version is released which needs only *lvgl* repository.

closed 2017-12-21 by kisvegabor

ajaybhargav, 2017-12-21:
that's awesome. Congratulations for reaching a milestone 👍