lvgl/lvgl #20: Allow background colour (mcolor) for labels

scopa90, 2017-08-28:
It might be good to allow the background colour of a label to be set via mcolor and inherit the parent object's mcolor if it's not supplied.

kisvegabor, 2017-08-28:
Hi,
it is a good idea, thank you! 
Do you mean if the label's style is NULL (inherited from parent) use transparent background and use _mcolor_ as background color if label's style is explicitly set?  Maybe the *empty* style attribute can disable this feature.

scopa90, 2017-08-28:
Or just keep it as transparent unless it's explcitly set?

kisvegabor, 2017-08-28:
Currently there is no tracking for a unique style attribute (like *mcolor*) whether it is set or not. Only I can see the entire style is inherited or not.

kisvegabor, 2017-08-28:
If we think it further the label can use all *backround related* style attributes like *mcolor*, *gcolor*, *bcolor*, *bopa* etc. but only if its style is set explicitly.  Else only use the inherited *font*, *ccolor*, *letter space* etc

kisvegabor, 2017-09-11:
I added the update to `beta`. Now if a label's style is set by *lv_obj_set_style()* to not NULL then background will be drawn. 
To avoid drawing background set *empty=1* and *bwidth=0* properties or copy LV_STYLE_TRANSP when initializing a new label style.

scopa90, 2017-09-16:
Just had chance to test this but only for a straight label.  Looks very good.

closed 2017-09-16 by scopa90

kisvegabor, 2017-09-16:
Thanks for testing!