lvgl/lvgl #21: Question about text area

scopa90, 2017-08-29:
I'm trying to use a text area which is one_line and pwd_mode both set true.  Using lv_ta_set_cursor_pos LV_TA_CUR_LAST and lv_ta_add_char to append characters shows them on the screen before they are replaced by '*'.  However, when I use lv_ta_get_txt I get 'Text area' returned and if I use lv_label_get_text I get an empty string.  If I also use lv_label_append_text then I am able to get the text back that I would expect.  Wouldn't it better to have lv_ta_get_txt return the contents of the text area or have I misunderstood how this works?

kisvegabor, 2017-08-29:
Hi,

it was really wrong. I fixed it an pushed the update in to `lvgl:beta` 

Please let me know how it is working for you now!

scopa90, 2017-08-29:
Fixed

closed 2017-08-29 by scopa90