lvgl/lvgl #22: Possible issue with buttons

scopa90, 2017-08-29:
Using default style for container and buttons (each button has its own object):

Before pressing "1":
[image]

After pressing "1":
[image]

Note corner "decorations" which appear after pressing

kisvegabor, 2017-08-29:
Hi,

It should work properly. Can you send your code?

scopa90, 2017-08-29:
This is the function which is used to create the buttons:

`#define PIN_X_PAD	5
#define PIN_Y_PAD	5
#define PIN_X_START	100
#define PIN_Y_START	200

lv_obj_t *bs[12];

void DrawKeyPad(lv_action_res_t (*kpFunc)(lv_obj_t *a, lv_dispi_t *b), lv_obj_t *scrn)
{
	uint16_t x, y;
	uint8_t c;
	char ltr[] = "1";
	lv_obj_t *label;

	x = PIN_X_START;
	y = PIN_Y_START;
	for(c=1; c!=10; c++)
	{
		bs[c] = lv_btn_create(scrn, NULL);
		lv_btn_set_pr_action(bs[c], kpFunc);
		lv_cont_set_fit(bs[c], true, true);
		lv_obj_set_pos(bs[c], x, y);
		label = lv_label_create(bs[c], NULL);

		ltr[0] = c+'0';
		lv_label_set_text(label, ltr);
		lv_obj_set_free_num(bs[c], c);

		if((c==3) || (c==6))
		{
			x = PIN_X_START;
			y = lv_obj_get_y(bs[c]) + lv_obj_get_height(bs[c]) + PIN_Y_PAD;
		}
		else
		{
			x = lv_obj_get_x(bs[c]) + lv_obj_get_width(bs[c]) + PIN_X_PAD;
		}
	}

	area_t coords;

	bs[0] = lv_btn_create(scrn, NULL);
	lv_btn_set_pr_action(bs[0], kpFunc);
	lv_cont_set_fit(bs[0], true, true);
	label = lv_label_create(bs[0], NULL);
	lv_label_set_text(label, "0");
	lv_obj_set_free_num(bs[0], 0);
	lv_obj_get_cords(bs[8], &coords);
	lv_obj_set_pos(bs[0], coords.x1, coords.y2 + PIN_Y_PAD);

	bs[10] = lv_btn_create(scrn, NULL);
	lv_btn_set_pr_action(bs[10], kpFunc);
	lv_cont_set_fit(bs[10], true, true);
	label = lv_label_create(bs[10], NULL);
	lv_label_set_text(label, "Done");
	lv_obj_set_free_num(bs[10], 10);
	lv_obj_get_cords(bs[3], &coords);
	lv_obj_set_pos(bs[10], coords.x2 + (PIN_X_PAD*3), coords.y1);

	bs[11] = lv_btn_create(scrn, NULL);
	lv_btn_set_pr_action(bs[11], kpFunc);
	lv_cont_set_fit(bs[11], true, true);
	label = lv_label_create(bs[11], NULL);
	lv_label_set_text(label, "Del");
	lv_obj_set_free_num(bs[11], 11);
	lv_obj_get_cords(bs[9], &coords);
	lv_obj_set_pos(bs[11], coords.x2 + (PIN_X_PAD*3), coords.y1);

}
`

kisvegabor, 2017-08-29:
I've tested you code it is rendered me correctly after pushing the button. The strange thing is the buttons have a radius by default which is missing on your image.  

Please, attache your `lv_conf.h`

[image]

scopa90, 2017-08-29:
lv_conf.h attached

[lv_conf.zip](https://github.com/littlevgl/lvgl/files/1260430/lv_conf.zip)

kisvegabor, 2017-08-29:
```c
/* Buffered rendering: >= LV_DOWNSCALE * LV_HOR_RES or 0 to disable buffering*/
#define LV_VDB_SIZE        (LV_HOR_RES * (LV_RES/10))		// (LV_HOR_RES * 30)
```
I suppose LV_RES is your own define. What is its value? Setting LV_RES to a 80 worked for me well.

scopa90, 2017-08-29:
LV_RES is set to the same as LV_VER_RES.  Setting it to 80 has fixed that issue but looks to have caused others but I'll dig into them and open a new issue if I can locate the cause.

kisvegabor, 2017-08-29:
It works for me with LV_VER_RES too.

It seems the borders are also missing on your buttons. Please try this code:
```c

static lv_style_t style_test;
lv_style_get(LV_STYLE_PRETTY_COLOR, &style_test);
style_test.empty = 1;
style_test.bwidth = 1 * LV_DOWNSCALE;
style_test.bcolor= COLOR_RED;
style_test.ccolor= COLOR_BLUE;

/*Red border*/
lv_obj_t * obj1 = lv_obj_create(lv_scr_act(), NULL);
lv_obj_set_style(obj1, &style_test);
lv_obj_set_pos(obj1, 10, 10);

/*Blue line*/
lv_obj_t * obj2 = lv_line_create(lv_scr_act(), NULL);
lv_obj_set_style(obj2, &style_test);
lv_obj_set_pos(obj2, 150, 10);
lv_line_set_auto_size(obj2, true);
static point_t points[2] = {{5, 5},{50, 5}};    /*Modify the coordinates to test*/
lv_line_set_points(obj2, points, 2);
```

It produces me the following result:
[image]

scopa90, 2017-08-29:
This is what I get:

[image]

scopa90, 2017-08-29:
One thought - I'm currently using beta code but have only been updating lvgl and not misc.  Does that need to be checked out as well?

kisvegabor, 2017-08-29:
Yes, if you use `lvgl:beta`, you should use `misc:beta` as well. But in your case it can not be the problem. Anyway please checkout please `misc:beta`

On your screenshot it is clearly seen the corners are not drawn. It is very strange. It seems the circle drawing algorithm is not working for you.  Please attache your *misc_conf.h* as well. Maybe we find the issue there.

Thank you in advance!

kisvegabor, 2017-08-29:
Please try this code as well to test circle drawing:
```c
point_t c;
cord_t tmp;
circ_init(&c, &tmp, 10);

cord_t x = 0;
cord_t y = 0;
while(circ_cont(&c)) {
    x = CIRC_OCT1_X(c);
    y = CIRC_OCT1_Y(c);
    printf("Circle x: %d, y: %d\n", x, y);    /*Somehow print/read the variables*/
    circ_next(&c, &tmp);
}  
```

For me the output is:
```
Circle x: 10, y: 0
Circle x: 10, y: 1
Circle x: 10, y: 2
Circle x: 10, y: 3
Circle x: 9, y: 4
Circle x: 9, y: 5
Circle x: 8, y: 6
Circle x: 7, y: 7
```

scopa90, 2017-08-30:
I get the same results for the code you gave above.  misc_conf.h attached
[misc_conf.zip](https://github.com/littlevgl/lvgl/files/1262798/misc_conf.zip)

kisvegabor, 2017-08-30:
misc_conf.h looks good.

Are you using compiler optimization? If yes please try without it.

If it doesn't help we need to dig deeper. In lv_draw/lv_draw.c   lv_draw_rect_border_corner() draws the rectangle border. You can dubug it to see whether it draws the circles cyclically or not. Don't care about what it does just see it draws in a cycle.

scopa90, 2017-09-01:
With optimisation off it creates such a large binary that it won't run.  This is no longer an issue, though, as I now have rounded corners on buttons and do not have the ghost border decorations.

kisvegabor, 2017-09-01:
I see. Then I close this issue again :)

closed 2017-09-01 by kisvegabor