lvgl/lvgl #24: password field not being updated correctly when lv_ta_del called

scopa90, 2017-09-03:
There looks to be an issue with a password text area field.  The following:

 ```
 lv_obj_t *pin_inp = lv_ta_create(lv_scr_act(), NULL);
  lv_ta_set_one_line(pin_inp, true);
  lv_ta_set_cursor_show(pin_inp, true);
  lv_ta_set_pwd_mode(pin_inp, true);
  lv_ta_set_text(pin_inp, "");
  lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
  lv_ta_add_char(pin_inp, '1');
  lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
  lv_ta_del(pin_inp);
  lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
  lv_ta_add_char(pin_inp, '2');
  char buff[3];
  strcpy(buff, lv_ta_get_txt(pin_inp));

```
Will leave "21" in buff instead of "2"

kisvegabor, 2017-09-03:
Hi,
Thank you for reporting the issue. I forget lv_ta_del() when fixed lv_ta_add_char/text()

I pushed the patch into the `beta` branch.

scopa90, 2017-09-03:
Fixed now. Thanks and enjoy your holiday.

closed 2017-09-03 by scopa90