lvgl/lvgl #26: About Text area (lv_ta) style

BenQoo, 2017-09-12:
I have some question about `lv_ta` on website's document

1. text style
    > _<http://www.gl.littlev.hu/object-types/text-area-lv_ta>_
    > ### Style usage
    > * ...
    > * Text style is inherited from the parent page.

   Dose the "parent page" mean `lv_ta`'s parent must be a `lv_page` object?
   But it seems that the text style can be accessed from label member in `lv_ta`.

2. cursor style
   >  ### Style usage
   > * ...
   > * Cursor width and color comes from the parent's ccolor and line_width style properties.

   I try to create a `lv_con` as the parent of `lv_ta`, then change its `ccolor` and `line_width`, but it doesn't apply to cursor.

Thank you!

kisvegabor, 2017-09-12:
Hi,
maybe I wasn't clear in the documentation. The Text area is a Page too and the Label inherits its style. 
For example:
```c
static lv_style_t style;
lv_style_get(LV_STYLE_PRETTY, &style);
style.ccolor = COLOR_RED;          //Red text
style.mcolor = COLOR_GREEN;        //Green color on the top of Text area

lv_obj_t * ta_1 = lv_ta_create(lv_scr_act(), NULL);
lv_obj_set_style(ta_1, &style);
```

BenQoo, 2017-09-13:
Hi,
Thanks for your reply.
I tested the following example, but `style.ccolor` not work on cursor and text.
The color of cursor and text is still in black.

```c
    static lv_style_t style;
    lv_style_get(LV_STYLE_PRETTY, &style);
    style.ccolor = COLOR_RED;          //Red text & red cursor?
    style.mcolor = COLOR_GREEN;        //Green color on the top of Text area
    style.line_width = 5 * LV_DOWNSCALE;

    lv_obj_t * ta_1 = lv_ta_create(lv_scr_act(), NULL);
    lv_obj_set_style(ta_1, &style);
    lv_obj_refr_style(ta_1);
```

[image]

BenQoo, 2017-09-13:
I have tried and found some ways to change the text color.
* To set a additional style object to label
* To change `ccolor` on the parent object of `lv_ta.

For example:
```c
    static lv_style_t style;
    lv_style_get(LV_STYLE_PRETTY, &style);
    style.ccolor = COLOR_RED;          // Red text
    style.mcolor = COLOR_GREEN;        // Green color on the top of Text area
    style.line_width = 5 * LV_DOWNSCALE;

    lv_obj_t * ta_1 = lv_ta_create(lv_scr_act(), NULL);
    lv_obj_set_style(ta_1, &style);

    /* create style for label */
    static lv_style_t txt_style;
    lv_style_get(LV_STYLE_PRETTY, &txt_style);
    txt_style.ccolor = COLOR_YELLOW;

    lv_obj_t * label = lv_ta_get_label(ta_1);
    lv_obj_set_style(label, &txt_style);
```
I'm not sure whether the two usages are correct or not.
Because it will become failed if I append the following function calls at the end of code.
```c
lv_obj_refr_style(ta_1);
```
or
```C
lv_page_set_sb_mode(ta_1, LV_PAGE_SB_MODE_OFF);
```

kisvegabor, 2017-09-13:
Sorry, I forget to say, already there is bugfix related to this. Please, check out lvgl and misc `beta`.
Your work around is okay too, but "one style" setting will work too in beta.

> I'm not sure whether the two usages are correct or not.
Because it will become failed if I append the following function calls at the end of code.
lv_obj_refr_style(ta_1);
or
lv_page_set_sb_mode(ta_1, LV_PAGE_SB_MODE_OFF);

What do you experience? `lv_obj_refr_style(ta_1);` is not required because `lv_obj_set_style()` do it as well.

BenQoo, 2017-09-13:
> What do you experience? lv_obj_refr_style(ta_1); is not required because lv_obj_set_style() do it as well.

I misunderstood that `lv_obj_set_style()` have to notify object the style is changed, so I added `lv_obj_refr_style()` after it. Thanks for your explanation, I get it!

After testing on `beta` branch, text portion is working fine now, the cursor width and color are not changed by `ccolor` and `line_width`.

kisvegabor, 2017-09-13:
Great! :)

`lv_obj_refr_style()` is required when your program is running and randomly you modify a style. The object need to be notified. However when you set  a new style with `lv_obj_set_style()` then the object will be refreshed to the new style automatically.

Thank you for reporting the issue. As it is working now I close it!

BenQoo, 2017-09-13:
sorry for my unclear.
On the cursor portion, I still can't control its width and color from `lv_ta`'s `ccolor` and `line_width`.

kisvegabor, 2017-09-13:
Sorry, I misread it. 
It was strange a behavior. I pushed a fix into `beta`. Now it also uses the Text area style like the text do.

kisvegabor, 2017-12-21:
Fixed in v5.0.0

closed 2017-12-21 by kisvegabor