lvgl/lvgl #28: Style for cursor

BenQoo, 2017-09-13:
Would it be possible to add additional cursor style to `lv_page` and `lv_ta`?

for example:
* block style
[image]
* block-outline style
[image]
* underline style
[image]

Thank you!

kisvegabor, 2017-09-13:
Good idea, and there is no obstacle to implement it. I will do it soon!
Thank you!

kisvegabor, 2017-09-13:
I started to implement this feature but some thing is not clear to me. How do these cursors should look like after the last character? (When you want to insert a char after last one)

kisvegabor, 2017-09-13:
This is the current state (for some reasons it is blinking on the gif...):

[image]

[image]

[image]

[image]

Any suggestions?

BenQoo, 2017-09-14:
Hi,
> I started to implement this feature but some thing is not clear to me. How do these cursors should look like after the last character? (When you want to insert a char after last one) . There is no character to be framed.
> Now I draw an empty frame or box like a terminal do?

It looks great! 👍 
I check the windows console and Visual Studio Code editor, they also place cursor at the last empty character. I think it's no problem in current state.

[image]
[image]
[image]

> Do you think is it required to specify cursor color different from text color?

I have a concern about the block style cursor.
If cursor and character take the same color, it will be unable to distinguish character whenever cursor pops up. It might be solved by:
1. Change text color to reverse color or different color whenever cursor pops up.
2. Specify a cursor color which is different from text color and background color as your current design.

I'm not sure which one is the better implementation for littleV, but it will be more flexible for customization if the cursor has its own color.

BenQoo, 2017-09-14:
Besides, I found two issue from the gif animation you posted.

1. The position of cursor is shifted left slightly.
Please refer to the following images:

   [image]  
   [image]

   [image]
   [image]

   [image]
   [image]

   [image]
   [image]

1. It seems the block cursor and block-line cursor don't popped up at whitespace character or it just time within the invisible period of cursor(?).

1. I recommend to make cursor animation restart whenever edit text or move cursor. It will keep the cursor displayed until user stop inputting.
  [image]
  [image]

kisvegabor, 2017-09-14:
Thank you for your feedback!

There was something bad with cursor position. Now I fixed it. 
I also reset the animation when a new cursor position is set.

Setting cursor color directly would be good ( e.g. lv_ta_set_cursor_color()) but I don't want to mix style properties with object attributes.  What do you think about set a cursor_style (lv_ta_set_cursor_style) to set:
* line_width: width in line, underline and outline mode
* mcolor: cursor colo
* ccolor: letter recolor in box mode

Adding a new styles is common among other object types too (like scroll bar style of page)

The current state is:
[image]

BenQoo, 2017-09-15:
Thanks for your reply!
I think it's no problem and the current state looks pretty nice! 👍

kisvegabor, 2017-09-16:
I pushed the new feature into the beta branch. Can you try it?

`lv_ta_set_cursor_type(ta, LV_TA_CURSOR_BLOCK/LINE/OUTLINE/UNDERLINE)` will modify the cursor type.

`lv_ta_set_cursor_style(ta, &style)` will set a new style.
* line_width: UNDERLINE and LINE width
* mcolor, gcolor: BLOCK, UNDERLINE, LINE color
* bcolor: OUTLINE color
* bwidth: OUTLINE width

By default the cursor style is create according to the label's style.

BenQoo, 2017-09-22:
Sorry for my late reply.
I will have time to test it next week.
Thanks for your update!

kisvegabor, 2017-09-23:
Okay, take your time! :)

BenQoo, 2017-09-25:
I found some issues and list as follows:

#### Line style
1. Cursor width and color will be overwritten by `bwidth` and `bcolor`
  [screenshot](https://user-images.githubusercontent.com/29389982/30791713-e968ee20-a1e6-11e7-9d22-c70634360ed8.png)
   ```C
    cursor_style.line_width = 3 * LV_DOWNSCALE; // UNDERLINE and LINE width
    cursor_style.mcolor = COLOR_RED;            // BLOCK, UNDERLINE, LINE color
    cursor_style.gcolor = COLOR_YELLOW;         // BLOCK, UNDERLINE, LINE color
    cursor_style.bcolor = COLOR_BLUE;           // OUTLINE color
    cursor_style.bwidth = 6 * LV_DOWNSCALE;     // OUTLINE width
   ```

#### UnderLine style
1. Same as _Line style_ issue 1.
1. Cursor draw abnormally at `\n` character. (only draw one pixel column)
   [screenshot](https://user-images.githubusercontent.com/29389982/30791760-5021f6fc-a1e7-11e7-8d3f-fea7398ece84.png)
   ```C
    cursor_style.line_width = 6 * LV_DOWNSCALE; // UNDERLINE and LINE width
    cursor_style.mcolor = COLOR_RED;            // BLOCK, UNDERLINE, LINE color
    cursor_style.gcolor = COLOR_YELLOW;         // BLOCK, UNDERLINE, LINE color
    lv_ta_set_text(ta_1, "123 56\nAB EFG");
    lv_ta_set_cursor_pos(ta_1, 6);
   ```
#### Outline style
1. Cursor draw abnormally at `\n` character.
    [screenshot](https://user-images.githubusercontent.com/29389982/30793281-69e001b8-a1f3-11e7-8171-f31d738b85cc.png)
   ```C
    cursor_style.bcolor = COLOR_BLUE;           // OUTLINE color
    cursor_style.bwidth = 6 * LV_DOWNSCALE;     // OUTLINE width
    lv_ta_set_text(ta_1, "123 56\nAB EFG");
    lv_ta_set_cursor_pos(ta_1, 6);
   ```

#### Block style
1. Cursor draw abnormally at `\n` character. (only draw one pixel column)

kisvegabor, 2017-09-25:
Hi,
thank you for this deep testing!

You are right with incorrect drawing on `\n`. I fixed it.

LINE, UNDERLINE and BLOCK style are drawn like a rectangle. So if you set a very thick border then it will occupy the whole space. Try `bwidth = 1 * LV_DOWNSCALE` on this styles. Sorry, it wasn't clear when I described the meaning of the style fields. So:
* line_width: UNDERLINE and LINE width
* mcolor, gcolor: BLOCK, UNDERLINE, LINE color
* bcolor: OUTLINE color *and BLOCK, UNDERLINE, LINE border color* 
* bwidth: OUTLINE width *and BLOCK, UNDERLINE, LINE border width*

I defined the style this way:
```c
static lv_style_t cursor_style;
lv_style_get(LV_STYLE_PLAIN, &cursor_style);
cursor_style.line_width = 3 * LV_DOWNSCALE; // UNDERLINE and LINE width
cursor_style.mcolor = COLOR_RED;            // BLOCK, UNDERLINE, LINE color
cursor_style.gcolor = COLOR_YELLOW;         // BLOCK, UNDERLINE, LINE color
cursor_style.bcolor = COLOR_BLUE;           // OUTLINE color and BLOCK, UNDERLINE, LINE border color
cursor_style.bwidth = 1 * LV_DOWNSCALE;     // OUTLINE color and BLOCK, UNDERLINE, LINE border color
cursor_style.ccolor = COLOR_LIME;           // BLOCK letter re-color
cursor_style.opa = OPA_50;                        //Cursor opacity
```

It looks like this from me right now:
[image]

BenQoo, 2017-09-26:
Hi,
Thanks for your explanation, I get it.

I have a additional question about UNDERLINE cursor.
Currently, UNDERLINE cursor increase the thickness toward the top(character) side.The room between UNDERLINE cursor and character will become narrow if chose a smaller font size.

[example](https://user-images.githubusercontent.com/29389982/30838863-b1edbca4-a2a0-11e7-863a-4e4826233731.png)

Such as the above screenshot, it only have 1 pixel height to increase UNDERLINE width.
Is it possible to adjust padding between them or make the thickness increased toward the bottom side?

Thank you!

kisvegabor, 2017-09-26:
It is a good point! A can adjust it equally up and down. 
Maybe the same thing should be done with LINE type but left and right directions.

kisvegabor, 2017-10-03:
Now I align LINE and UNDERLINE equally top-bottom/left-right.

BenQoo, 2017-10-05:
I found a concern about d6ccc48963343119708b032d046d54db9bf4ff68.
To increase the width toward the both side, It will have the same problem as the original design when set a thicker underline cursor.

I have an idea which is to take cursor style's `vpad` and `hpad` to adjust the cursor position, 
and we don't have to consider the directions for width increasing.
What do you think?

```C
if (ta_ext->cursor_type == LV_TA_CURSOR_LINE) {
    cur_area.x1 = letter_pos.x + ta_ext->label->cords.x1 + cur_style.hpad;
    cur_area.y1 = letter_pos.y + ta_ext->label->cords.y1 + cur_style.vpad;
    cur_area.x2 = letter_pos.x + ta_ext->label->cords.x1 + cur_style.hpad + cur_style.line_width;
    cur_area.y2 = letter_pos.y + ta_ext->label->cords.y1 + cur_style.vpad + letter_h;
    lv_draw_rect(&cur_area, mask, &cur_style);
else if (ta_ext->cursor_type == ...) {
    ...
} else if (ta_ext->cursor_type == LV_TA_CURSOR_UNDERLINE) {
    cur_area.x1 = letter_pos.x + ta_ext->label->cords.x1 + cur_style.hpad;
    cur_area.y1 = letter_pos.y + ta_ext->label->cords.y1 + cur_style.vpad + letter_h - cur_style.line_width;
    cur_area.x2 = letter_pos.x + ta_ext->label->cords.x1 + cur_style.hpad + letter_w;
    cur_area.y2 = letter_pos.y + ta_ext->label->cords.y1 + cur_style.vpad + letter_h;
    lv_draw_rect(&cur_area, mask, &cur_style);
}
```

kisvegabor, 2017-10-09:
Added as you have suggested.
In addition OUTLINE and BLOCK size will increased with +-hpad/vpad.

BenQoo, 2017-10-16:
Hi,
There is an issue about outline cursor, `vpad` didn't change the size, it change the position.

Besides, the line of border is blurred at some edge, is it the expected rendering?
[image]

kisvegabor, 2017-10-16:
Hi,
a sign was bad. I fixed it in `beta`.

The blur is the expected behavior because of anti aliasing. When a 1 px wide line should be drawn between two pixels then two lighter line will appeare.

BenQoo, 2017-10-17:
Thanks for your reply, I get it.

`vpad` on outline cursor is working now. :)

kisvegabor, 2017-10-17:
It seems I'm not so good at testing things very well. :) Therefore I decided to create test codes for all object types. Unfortunately it can happen that a new feature breaks an old or "something" happens during merge. All test cases will contain images too with the expected result.

It will be launched in the next released. The current state is here: https://github.com/littlevgl/examples/tree/dev-5.0/lv_tests

BenQoo, 2017-10-17:
👍 

However, I think it's common that it could meet some issues or bugs while developing a new feature. so don't take that too much to heart. :)

kisvegabor, 2017-11-20:
I released the beta version of v5.0 (lvgl/beta, misc/beta, proj_pc/beta).
As it is major version it is not backward compatible. (mainly because of the styles)
This small code shows the new style usage system:
```c
    static lv_style_t ta_cur_style;
    lv_style_copy(&ta_cur_style, &lv_style_pretty);
    ta_cur_style.body.main_color = COLOR_RED;

    lv_obj_t *ta = lv_ta_create(lv_scr_act(), NULL);
    lv_ta_set_cursor_type(ta, LV_CURSOR_BLOCK);
    lv_ta_set_style(ta, LV_TA_STYLE_CURSOR, &ta_cur_style);
```
I hope you can make some test to approve this feature in v5.0

kisvegabor, 2017-12-21:
Hi,

thank you for raising the idea and for testing.
This feature is finally released in  v5.0.0

closed 2017-12-21 by kisvegabor