lvgl/lvgl #29: Question on color16_t

scopa90, 2017-09-13:
Looking in the source I can see that color16_t is using uint16_t as opposed to uint8_t.  Is there a reason for this?

kisvegabor, 2017-09-13:
I'm not sure I understand well but color16_t is for 16 bit color depth. To store this a 16 bit variable is needed like uint16_t

scopa90, 2017-09-13:
Please let me phrase it better!  The struct that makes up color16_t has uint16_t for red, green, blue.  Shouldn't they be uint8_t ?

kisvegabor, 2017-09-13:
color16_t looks like the following. 

```c
typedef union
{
    struct
    {
        uint16_t blue  :5;
        uint16_t green :6;
        uint16_t red   :5;
    };
    uint16_t full;
}color16_t;
```
The `:5`, `:6` and `:5` in the *struct* means it is bit field. So red 5 bit, green 6 bit, blue 5 bit. The whole struct is 16 bit.

Because _colo16_t_ is an union, *full* occupies the same space as the *struct*. This way the color can be read as a number via *full* or the components can be set easily with the *struct*.

scopa90, 2017-09-13:
Thanks for explaining that.

closed 2017-09-13 by scopa90

kisvegabor, 2017-09-13:
Your welcome!