lvgl/lvgl #30: Bigger free_num for lv_obj_t?

scopa90, 2017-09-14:
At present I'm using the free_num in the returned object to add an index on listview items but envisage that there will be more than 1000 items.  Would it be possible to add a free_word or is there a better way to store the index values?

kisvegabor, 2017-09-14:
I can modify the `LV_OBJ_FREE_NUM` define in lv_conf.h to set the the type of free_num. For example:

`#define LV_OBJ_FREE_NUM   uint16_t`

Using 1000  element is list could cause other issues. If an element is e.g. 50 px heigh, the last will be on 50000 px coordinate. Now the GUI should be between -16K and +16K. See *misc/gfx/area.h*

scopa90, 2017-09-14:
Interesting point.  If I limit it to 300 then that would be enough (although still outside the 256 range for 8 bits).  Either that or I'm more intelligent in how I load the list into memory and the listview.

kisvegabor, 2017-09-14:
If you tell what you would like to do hopefully I can suggest a solution. (in a new topic)

scopa90, 2017-09-14:
Thanks for the offer.  Let me play about with it here and see what I can come up with first as I don't want to waste your valuable time (which should be devoted to further develping littlevgl!)

kisvegabor, 2017-09-14:
Okay, se what happens :)
Anyway, the updating of `LV_OBJ_FREE_NUM` still might be a useful thing.

kisvegabor, 2017-11-20:
In `lv_conf.h` I added  `LV_OBJ_FREE_NUM_TYPE` to define the type of the free number (e.g. uint16_t)

It is available in `lvgl/beta`. It is the beta version of v5.0 so there are non-backward compatible API changes in it.

kisvegabor, 2017-12-21:
Added in v5.0.0

closed 2017-12-21 by kisvegabor