lvgl/lvgl #31: How to use UTF-8 encoding?

abauras, 2017-09-15:
Thanks for your work. Please tell me how I can use fonts with UTF-8 encoding support

kisvegabor, 2017-09-15:
Hi,
unfortunately currently there is no UTF-8 support in the library. However these characters are available: http://www.utf8-chartable.de/

Is it help you or do you need characters from full UTF-8 character set?

abauras, 2017-09-15:
Thanks for the support. I need to use Cyrillic and Latin characters together in one text (and in one fonts), i.e. source code of the program in UTF-8 encoding.

kisvegabor, 2017-09-15:
UTF-8  support would be great in the library so the current font handling method could be ungraded.

I have an ide how to do it easily. However I can deal with it only on next week. I hope you are not in a hurry. :)

My idea is to update *lv_label* to handle multiple fonts, and according to the UTF-8 character code it chooses the right font.

abauras, 2017-09-15:
I'm in no hurry :) I hope for the successful implementation of your idea, which will help the use of multilingual text

kisvegabor, 2017-09-23:
I'm doing well with UTF-8 support. Now the library is able to render and handle UTF-8 coded characters. 
Still the font converter scripts needs some update. On next week you can try this new feature! :)

kisvegabor, 2017-09-27:
Hi,
good news here! I pushed the UTF-8 update to  `lvgl:utf8` and `misc:utf8`

Please update your the **FONT** and **TEXT** section of your `misc_conf.h` from `misc_conf_templ.h` 

You will find a lot of fonts which can be enabled. E.g.: `USE_FONT_DEJAVU_20_CYRILLIC      1`

In `misc_conf.h` you need to enable `TXT_UTF8    1`

I'm waiting for your feedback!

abauras, 2017-10-03:
Support for fonts UTF-8 works fine. Thanks for your work. I'm looking forward to version 5

closed 2017-10-03 by abauras

kisvegabor, 2017-10-03:
Great!
Version 5 will be released in a few weeks.

KcMeterCEC, 2017-10-10:
Hi kisvegabor, 

I found the struct `font_t` has been changed.

So, the **font file**  who have generated by  python script **fnt2c.py** have to be change too.

But I have not idea about how to change the script file to generate **the start index of the glyphs in the bitmap array**.

I think you have finished the script file, do you ?
Thanks!

kisvegabor, 2017-10-10:
As the UTF-8 support is not released yet all the related things are in the `beta` branches. (lvgl/beta, misc/beta, utils/beta) That is why there is no documentation about it yet. No not forget to update the TEXT and FONT section of misc_conf.h from misc_conf_templ.h

Here is how does it work:
- Generate an fnt and png file with BMFont (It doesn't changed) (http://www.gl.littlev.hu/font-import)
- Use the `fnt2c.py` from utils/beta (`fnt2c-py -h` to see the usage)
- In misc_conf.h enumerate your new font names in FONT_NEW_NAMES
- Add the new fonts with: `font_add(MY_FONT_NAME, my_font_get_dsc, parent_font)`   
  - After that MY_FONT_NAME will be handled together with the parent font and the library will look for glyphs in "children" fonts too.
  - E.g. `font_add(ARIAL_10_BOLD_GREEK, arial_10_bold_greek_get_dsc, font_get(ARIAL_10_BOLD)`
  - When creating a base font (e.g. ARIAL_10_BOLD) set *parent font* to *NULL*
- Later just set the base font as font. E.g. `style.font = font_get(ARIAL_10_BOLD)`
- See font.c for examples

There are a lot built-in fonts in the library. You just need to enable them in *misc_conf.h*

Do not hesitate to ask me if something doesn't work.