lvgl/lvgl #32: Empty lv_list ignores lv_obj_set_size

scopa90, 2017-09-21:
Using the following code:

```
panelId = lv_list_create(sscrn, NULL);
lv_obj_set_size_us(panelId, LCD_GetXSize()-20, LCD_GetYSize() - 135);
lv_obj_set_pos_us(panelId, 10, 10);
```

I find that if the lv_list is empty then the panel size is a single line, i.e. it ignores the set_size.  Is that deliberate?

kisvegabor, 2017-09-21:
Lists are Pages inside. The Pages have two parts parts: a background and a scrollable part on it.  The list elements are on the scrollable part. 

By default the background's style is LV_STYLE_TRANSP. So when you set the the list's size you set this transparent background object. 

It is the intended behavior because this way you can set the maximum size occupied by the list. If you have more list elements then the list can be scrolled.

Use `lv_obj_set_style(panelId, lv_style_get(LV_STYLE_PRETTY_COLOR, NULL))` to set a non-transparent style for the background.

scopa90, 2017-09-21:
Thanks for your answer. That covers everything I need to know on this topic.

closed 2017-09-21 by scopa90