lvgl/lvgl #33: Changing colour of buttons in lv_list

scopa90, 2017-10-01:
Looking for some advice.  I have a populated lv_list and would like the button to permanently change colour if it is selected and switch back to the "normal" colour if another button is then selected (with the new button changing colour permanently). 

I can get the colour to change by attaching a function to the release event and, in that function, doing an lv_btn_get_style on the passed lv_obj_t and then updating the ccolor and mcolor.  I then store the passed lv_obj_t to change the colour back on the next button press.  The first press works but the colour change on the next button doesn't reset the first button.  Is there a way to do this?

kisvegabor, 2017-10-02:
Hi,
the problem is all of your buttons use the same style. If you modify that style then all the button's style will be modified.  (Because objects only store a pointer to the style)

My **suggestion** is to use the **toggle attribute** of the buttons. 
Enable it with `lv_btn_set_tgl(btn, true)`. In this case after release the button goes to toggled state with a different style. 

With `lv_btn_set_styles()` set the styles in each state. In the release action you can modify the state of the "old" button with `lv_btn_set_state(btn_prev, LV_BTN_STATE_REL)`.

Does it working?

scopa90, 2017-10-03:
That works perfectly. Thanks for that.

closed 2017-10-03 by scopa90

kisvegabor, 2017-10-03:
Great!