lvgl/lvgl #34: Symbol font update

kisvegabor, 2017-10-02:
I've optimized the font storage to use less memory. 
@ajaybhargav please generate the symbol fonts again with the new python script: https://github.com/littlevgl/utils/blob/beta/font_create/fnt2c.py

Usage: 
`python fnt2c.py -f <font_name> [-o <output file> -s <start unicode> -e <end unicide>]`

Or see `python fnt2c.py -h` for a detailed explanation.

If you have no time right now to deal with it just send me the fnt and png files and I will do this.

Thank you in advance!

ajaybhargav, 2017-10-02:
Hi Gabor,
I am attaching Symbol_30 and Symbol_60 fnt + png files. I have got lot of work piled up due to holidays and it doesn't seem like ending soon.

Let me know if you need anything else.

[font_create.zip](https://github.com/littlevgl/lvgl/files/1348338/font_create.zip)

[image]

[image]

ajaybhargav, 2017-10-02:
I am also attaching config files used for creating fonts. So it will be easier for you.

[font_create_config.zip](https://github.com/littlevgl/lvgl/files/1348343/font_create_config.zip)

kisvegabor, 2017-10-02:
Thank you very much! 
And good luck in your work!

closed 2017-10-02 by kisvegabor