lvgl/lvgl #36: lv_ta password issue

scopa90, 2017-10-05:
The following code should leave "1234" in buff but actually leaves "***4":

` lv_obj_t *pin_inp = lv_ta_create(lv_scr_act(), NULL);
 lv_ta_set_one_line(pin_inp, true);
 lv_ta_set_pwd_mode(pin_inp, true);
 lv_ta_set_text(pin_inp, "");
 lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
 lv_ta_add_char(pin_inp, '1');
 lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
 lv_ta_add_char(pin_inp, '2');
 lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
 lv_ta_add_char(pin_inp, '3');
 lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
 lv_ta_add_char(pin_inp, '5');
 lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
 lv_ta_del(pin_inp);
 lv_ta_set_cursor_pos(pin_inp, LV_TA_CUR_LAST);
 lv_ta_add_char(pin_inp, '4');
 char buff[5];
 strcpy(buff, lv_ta_get_txt(pin_inp));`

kisvegabor, 2017-10-05:
It was already fixed but since then there was a lot of updates on Text Area and somehow it was overwritten. Now I cleaned the branches. I hope it won't happen again. Or at least not soon :)

Please update misc to `beta` too.

Now the UTF-8 support is merged to `beta`. It might be useful for you. Please. check the FONT and TEXT section of `misc_conf_templ.h` to see the news.

scopa90, 2017-10-05:
Updated to beta and it's all fixed now.
UTF8 support is a good move!

closed 2017-10-05 by scopa90