lvgl/lvgl #37: Is it possible to set z-order of objects?

scopa90, 2017-10-05:
I have the following sequence of events:  Create list full width of screen -> create smaller list -> create screen 2 which covers the physical screen -> delete full width screen list -> delete screen 2.  Now the smaller is list is below the full width list.  I can see why this is happening (order based on creation time) but I wondered if it is possible within the existing code to change the order so that the smaller list always stays on top of the full width list?

kisvegabor, 2017-10-05:
So:
1. Create a screen wide list
2. Create a small list
3. Create an other screen (screen 2) And not load it (I suppose)
4. Delete  screen wide list
5. Delete screen 2

>Now the smaller is list is below the full width list

How is possible if you deleted the full width list?

Anyway, if you apply `lv_obj_set_parent(obj, act_parent)` (so set its current parent again) it will be moved to the top.

scopa90, 2017-10-05:
Sorry, missed a step:

4.5 Create screen wide list.

Thanks for the pointer to lv_obj_set_parent, That's what I was looking for.

closed 2017-10-05 by scopa90