lvgl/lvgl #38: Text poistion in lv_ta

BenQoo, 2017-10-06:
I have a question. If I would like to adjust the position of text in `lv_ta`, which `vpad` and `hpad` should I take?
* Style of `lv_ta`
* Style of scrollbar `style_sb` from `lv_page_get_style_sb(ta)`

I made some try. I use ptask to change vpad periodically.
I set `vpad` on style of `lv_ta`, it worked on the first time call `lv_obj_set_style(ta, &ta_style)`, but it didn't work on the second time or set style to the other new `lv_style_t`. 
About `style_sb`, It's no effect by changing its `vpad`.

Thanks!

kisvegabor, 2017-10-06:
It really doesn't work properly. I will fix it on Monday. Until that please try this: `lv_obj_set_style(lv_page_get_scrl(ta), &style)` it will set the style of the transparent scrollable part of text area. Use LV_STYLE_TRANSP_TIGHT as starting point.

vpad and hpad now do not affect the scrollbar position but it is a good idea to fo that!

kisvegabor, 2017-10-09:
Now the label's position should be affected by hpad and vpad. 
Can you test it?

I will add the scrollbar part soon.

BenQoo, 2017-10-16:
> Now the label's position should be affected by hpad and vpad.
> Can you test it?

No problem.
After testing, It works now, thank you!

> How do you imagine the scrollbar and hpad/vpad relationship?
Sorry, I'm not familiar with them. I guess it can change hpad/vpad to adjust the position of scrollbar.

kisvegabor, 2017-10-16:
I'm happy that it's working! :)

I will figure out something for scrollbar.

kisvegabor, 2017-11-20:
In beta scrollbar positions are affected by `style.body.padding.hor` and `style.body.padding.hor`.
(Please update you `misc_conf.h` and` lv_conf.h` from `misc_conf_templ.h` and` lv_conf_templ.h`)

Like this:
```c
    static lv_style_t sb_style;
    lv_style_copy(&sb_style, &lv_style_plain_color);
    sb_style.body.padding.hor = -15;
    sb_style.body.padding.ver = 15;

    lv_obj_t *page = lv_page_create(lv_scr_act(), NULL);
    lv_page_set_style(page, LV_PAGE_STYLE_SB, &sb_style);
    lv_page_set_sb_mode(page, LV_SB_MODE_ON);
    lv_page_set_scrl_fit(page, true, true);
    lv_obj_set_size_scale(page, 100, 200);
    lv_obj_set_pos_scale(page, 10, 10);

    lv_obj_t *label = lv_label_create(page, NULL);
    lv_label_set_text(label, "long long lines\n\n"
                             "to shows the\n\n"
                             "scrollbars'\n\n"
                             "position.\n\n"
                             "Some dummy text\n\n"
                             "Some dummy text\n\n"
                             "Some dummy text\n\n");
```
[image]

kisvegabor, 2017-12-21:
Added in v5.0.0

closed 2017-12-21 by kisvegabor