lvgl/lvgl #39: Suggestion for change to lv_btn_set_styles

scopa90, 2017-10-07:
First of all I am working with the latest beta and can see many many improvements.  Two immediate changes are that fonts are clearer and the code seems smaller so thank you for that.

Would it be an idea to allow lv_btn_set_styles to use NULL for button styles which are not to be changed and just supply the ones you want to alter? Or is there already a function to do this?

kisvegabor, 2017-10-08:
I'm glad to hear that you like the new improvements.

It's a good idea to leave the style of given state unchanged when set to NULL.
  
I have a constructive idea. Because it is super easy to implement in `lv_btn_set_styles` function I suggest that add this improvement yourself and then create a pull request on GitHub. It would be good for both of us because an extra contribution is always welcome! :)

Please quickly read this document before modyfing code: https://github.com/littlevgl/lvgl/blob/master/docs/CONTRIBUTING.md#formatting

scopa90, 2017-10-08:
Oddly enough I've made the changes in my copy and was going to send over a small patch file.  The only possible issue with the pull request is the include paths - if you recall I've mentioned this in the past and my copy is setup for my project with misc and misc under lvgl. Let me read through the doc you've linked to and I'll see if it's possible.

kisvegabor, 2017-10-08:
I see. Maybe try this to commit only specific parts of a file: https://stackoverflow.com/questions/1085162/commit-only-part-of-a-file-in-git
It might be useful later too.

Anyway I wouldn't like to cause to much pain with this. If it's too difficult to contribute in this way I can also add the update.

closed 2017-10-09 by scopa90