lvgl/lvgl #42: Avoid abbreviations in API?

kisvegabor, 2017-10-19:
The next release will be a major release therefore it is possible to update API.
The question is: **is it good to abbreviate in API function's name or not?**
For example:

|           Current                      |                         Idea                              |
| ------------------------------------ |----------------------------------------------------|
| `lv_obj_set_pos()`               |      `lv_object_set_position()`                 |
| `lv_btn_set_state()`             |      `lv_button_get_state()`                     |
| `lv_ddlist_set_options()`      |      `lv_dropdownlist_ set_options()`      |
| `lv_ta_set_cursor_pos()`     |      `lv_textarea_set_cursor_position()`  |

The not abbreviated names are more descriptive but takes more time to write and read.
What's worth more? What is you opinion?

scopa90, 2017-10-23:
My opinion is stay with the shortened names.  They're quicker to type and are readable.

abauras, 2017-10-23:
I like the option with full function names, although they can be implemented by defining abbreviated names

kisvegabor, 2017-10-23:
Meanwhile I tried different abbreviation "styles":
In my opinion the goal is to provide names which are so descriptive that you can know what it will do without (near) any prior knowledge.
For example the current `LV_BTN_STATE_TPR` (toggled pressed state of button) is obscure. `LV_BTN_STATE_ON_PRESSED` would be better.

In general keeping modul names (like btn, ta, ddlist) short but write longer "description part" seems a good choice. Like  change `lv_btn_set_tgl()` to `lv_btn_set_toggle()`. Hence you have to type only a few letters to get relevant result from auto-complete in your IDE and then the results will be very clear. This way not all off the current function names become obsolete (e.g. lv_button_set_toggle style would crash all the names) only a few updates will occur. 

Do you agree?

scopa90, 2017-10-23:
So long as you provide a way (by defines or other) so that existing code can compile cleanly then it's not really an issue.  You'd be better asking anyone who has no knowledge of littlevgl what their opinion is, I think.

kisvegabor, 2017-10-23:
Until it is possible I will ensure backward compatibility. However unfortunately I can't ensure that the API will never change and will be backward compatible as the library evolves. 
Only major releases (like the following v.5.0.0) will contain API modifications. The minor releases (like 4.1.0, 4.2.0, 4.3.0) are always backward compatible. 

I know it is a pain for you and I feel sorry for that. But sometimes it is the price of introduce better things or improve the current state.

scopa90, 2017-10-24:
Fully understand that and I'll take a look at the release notes for v5 and see if there's anything I'll need. If not then I'll just stay on my current version which works really well.

kisvegabor, 2017-10-24:
Okay! :)

[Here](https://github.com/littlevgl/lvgl/tree/master/docs) in TODO files you can always keep track with the new features for the new releases.

closed 2017-11-20 by kisvegabor