lvgl/lvgl #43: Send signal to object without display input

BenQoo, 2017-10-20:
Is it possible to send signal with `NULL` display input?

I would like to sync the behavior of a physical button with a `lv_btn`, so I try to send signals to it.
```C
lv_btn_signal(btn, LV_SIGNAL_RELEASED, NULL)
```
In current implementation, to set display input to `NULL`, it will cause the following function error when button is released.
https://github.com/littlevgl/lvgl/blob/c96b1d4db37a2ab3156f7a6f97f6eed2052dfaca/lv_obj/lv_dispi.c#L113

kisvegabor, 2017-10-20:
It is recommended to use the button as a display input. 
In `hal/indev.c` add your button handle as a second input device

```c
bool indev_get(uint8_t indev_id, int16_t * x, int16_t * y)
{
  if(id == 0) return toucpad_get(x, y);
  else if(id == 1) return button_get(x,y);    /*Return true if the button is pressed; false if released*/
}
```
Your `button_get` function could look like this:
```c
bool button_get(int16_t * x, int16_t * y)
{
    bool pressed = false;

    if(button_is_pressed()) {
        *x = lv_obj_get_x(my_lv_btn) + 1;
        *y = lv_obj_get_y(my_lv_btn) + 1;
        pressed = true;
    }

    return pressed;
}
```

In `indev.h` set `#define INDEV_NUM 2`

To be honest I haven't tried it yet (I mean the multiple indev handling) but is should work.

BenQoo, 2017-10-25:
Thanks for your reply, I get it.
I will think it through this portion.

closed 2017-10-25 by BenQoo