lvgl/lvgl #44: Multitouch Support

tjstyle, 2017-11-18:
Some LCD display with capacitive touch have multi touch capability.

like for STM32, commonly typedef struct used is like this : 
**typedef struct
{
  uint8_t  touchDetected;                /*!< Total number of active touches detected at last scan */
  uint16_t touchX[TS_MAX_NB_TOUCH];      /*!< Touch X[0], X[1] coordinates on 12 bits */
  uint16_t touchY[TS_MAX_NB_TOUCH];      /*!< Touch Y[0], Y[1] coordinates on 12 bits */

#if (TS_MULTI_TOUCH_SUPPORTED == 1)
  uint8_t  touchWeight[TS_MAX_NB_TOUCH]; /*!< Touch_Weight[0], Touch_Weight[1] : weight property of touches */
  uint8_t  touchEventId[TS_MAX_NB_TOUCH];     /*!< Touch_EventId[0], Touch_EventId[1] : take value of type @ref TS_TouchEventTypeDef */
  uint8_t  touchArea[TS_MAX_NB_TOUCH];   /*!< Touch_Area[0], Touch_Area[1] : touch area of each touch */
  uint32_t gestureId; /*!< type of gesture detected : take value of type @ref TS_GestureIdTypeDef */
#endif  /* TS_MULTI_TOUCH_SUPPORTED == 1 */

} TS_StateTypeDef;**

So it's good if multi touch capability can be implemented in lvgl.

kisvegabor, 2017-11-20:
Hi,

can you mention a few examples where the multi touch support could be useful on embedded systems?

tjstyle, 2017-11-23:
About multitouch I think that not really needed in lvgl usability.
Because there is no need to zooming image, multi hold keyboard, and painting in lvgl.
So I close this enhancement suggestion.

closed 2017-11-23 by tjstyle