lvgl/lvgl #45: muti language support

helezh, 2017-11-18:
how to support muti language,such as Chinese Japanese and Korean

kisvegabor, 2017-11-20:
Hi,

The new version of LittlevGL will support UTF-8. I just released a beta version of it.   
See lvgl/beta, misc/beta, utils/beta

Keep in mind this is major release so it is not compatible with v4.2. So the first thing is to update your `misc_conf.h` from `misc_conf_templ.h` and enable TXT_UTF8

So how to use UTF-8 chacters:
 * Generate an fnt and png file with BMFont (It doesn't changed) (http://www.gl.littlev.hu/font-import)
 * Use the fnt2c.py from utils/beta: `python fnt2c.py -f my_example -s first_uncode_id -e last_unicode_id
 * Copy the generated files into your project and enable the new font in misc_conf.h: `#define USE_FONT_MY_EXAMPLE   1`
 * Use the font in a style: `style.text.font  = &font_my_example` (include my_example.h too)
 * You can create a "font pack" with `font_add(&font_base, &font_extension)` This way you have to set font_base as font, but  the library will look for letter maps in the added fonts too.

Still there are no documentation about the new version but it is  mainly compatible with the old except the styles and some function names. 

Do not hesitate to ask me if something doesn't work.

helezh, 2017-11-24:
I used  your new lvgl/beta and misc/beta in "lv_proj_pc_v4_2_0" project，build successful but runing tft simulator is blank,why?
1,I download the code is lvgl-beta.zip and misc-beta.zip,from your github.
2,removed the files in "lv_pro_pc_v4_2_0/misc/*" and "lv_pro_pc_v4_2_0/lvgl/*"
3,unzip  lvgl-beta.zip and misc-beta.zip to lv_pro_pc_v4_2_0/misc/ and lv_pro_pc_v4_2_0/lvgl/.
4.make new lv_conf.h and misc_conf.h. and masked //#if 0           //#endif"
5.make main.c, the code is 
"
#include <stdlib.h>
#include <unistd.h>
#include <stdio.h>
#include "hw/hw.h"
#include "misc/misc.h"
#include "misc/os/ptask.h"
#include "lvgl/lvgl.h"
//#include "lvgl/lv_app/lv_app_util/lv_app_notice.h"

int main (void)
{
    /*Initialization*/
    misc_init();
    per_init();
    dev_init();
    lv_init();
   lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);

    while(1) {
        /* Periodically call the ptask handler.
         * It could be done in a timer interrupt or an OS task too.*/
        ptask_handler();
        usleep(5000);
    }

    return 0;
"
6,build success,but running is blank,how can i do? thank you!

kisvegabor, 2017-11-24:
Please, use `proj_pc/beta`.

Because HAL is integrated in LittlevGL, in v.5.0 you have to initialize it by setting up an `lv_disp_drv_t` structure. It needs the same function (disp_map, disp_fill) like `hal` before.

You also have to call `lv_tick_handler()` in every millisecond.

helezh, 2017-11-27:
the v5 is unfinished? many of function can not to use. and i found of   main.c/116(" indev_drv.name = "Mouse";") has a problem, i correct lv_hal_indev.h/lv_indev_drv_t(add uint8_t *name).

kisvegabor, 2017-11-27:
No, it is still under development. 
Now there are 3 relevant branches: master (previous v4.2), beta (v5.0 "preview"), dev-5.0 (for development) In proj_pc you find the same beanches with a working initialization. Please refer to it. ("name" was intentionally removed from lv_disp_drv_t)

So v5.0 is not stable yet. I hope it will be in one or two weeks.

kisvegabor, 2017-11-27:
The good thing is the last major improvement was implemented yedterday . It was integrating misc, lvgl and hw into 1 repo.
Only some minor renames and testing remains.

kisvegabor, 2017-12-21:
Hi,

v5.0.0 is released now with Unicode support. Please check the [Documentation](http://www.gl.littlev.hu/basics)

closed 2017-12-21 by kisvegabor