lvgl/lvgl #46: How to support the right-to-left or bi-directional text.

jlwg, 2017-11-22:


kisvegabor, 2017-11-22:
Hi,
currently there is no support for this feature. 

Does it mean when you use:
`lv_label_set_text(label, "ABC")`

it should write:
`                "CBA"`
and also flush text to the right?

jlwg, 2017-11-23:
yes ,for example :  Arabic scripts are
written from the right to the left and that it does not know uppercase and lowercase
characters. Further the character codes of the text are not identical with the character index in the font file used to render the character, because the notation forms of
the characters depend on the positions in the text。

jlwg, 2017-11-23:
some language is  right-to-left  ,for example : Arabic scripts are
written from the right to the left and that it does not know uppercase and lowercase
characters. Further the character codes of the text are not identical with the character index in the font file used to render the character, because the notation forms of
the characters depend on the positions in the text。

kisvegabor, 2017-11-23:
I can make the library to support drawing the letters in a revers order. 

>  the notation forms of the characters depend on the positions in the text。

Can you, please, show a real life example?

kisvegabor, 2017-11-24:
I found this: https://en.wikipedia.org/wiki/Arabic_script_in_Unicode#Contextual_forms
Do you think something like this?

If yes, then this kind of mapping is implementable.

However it means a lot of characters. Is it serious issue in Arabic language if you let the "base" latter when the substitute is not available?

jlwg, 2017-11-25:
you can download the pdf file from the site:  https://www.segger.com/downloads/emwin/UM03001  
and look up the "chapter 32 language support--->32.3 Right to left- and bidirectional text"

kisvegabor, 2017-11-25:
Now its clear. Thank you! 
We need a function which "translates" the letters according to its position.
Can help with implementing this function? And I will do the rest.

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor