lvgl/lvgl #47: How to rotate the screen

mithunkamat, 2017-12-01:


kisvegabor, 2017-12-01:
Hi,

you need rotate the screen in you display driver. The graphics library just know the resolution. It depends on you how to put the pixels into your frame buffer.

mithunkamat, 2017-12-01:
Can we do it runtime? How?
Please help.

kisvegabor, 2017-12-01:
Now you can't rotate the display in run-time because LV_HOR_RES and LV_VER_RES are defined in build-time.  

Without modifying the library you can do the following:
1. Let's say you have got 240x320 display
2. Define LV_HOR_RES 320 and LV_VER_RES 320 (so as square)
3. When you create a screen add a transparent object on it and put all of your object to it.
4. Set the transparent object's size to 240x320 or 320x240 according to the orientation
5. Also change your frame buffer write algorithm according to the orientation (rows first or cols first)

Or simply as in step 2 but create your object on the screen and rearrange them when the display turned

closed 2017-12-06 by mithunkamat

hydra, 2018-07-24:
See #329