lvgl/lvgl #48: can  littlevgl support movies?

jlwg, 2017-12-02:
I want it to play video files, so i want to Make a mediaplayer by littlevgl .

kisvegabor, 2017-12-02:
No, there is no interface for that and no codecs.

However if you write the codecs(decoding) I can add a canvas or to put the images there.

jlwg, 2017-12-02:
yes ,i need this canvas to provide a mechanism  for putting the images dynamically.

kisvegabor, 2017-12-02:
I think it is not necessary to allocate memory for a canvas just to set an area which won't be used by LittlevGL. This way you can put pixels directly into the frame buffer. The only small restriction is you can't put buttons on the movie area but you can below the it.

What do you think?

Matheus-Garbelini, 2017-12-24:
@kisvegabor i think that's a great idea, as for the buttons i think the ideal is to put them below the video frame. I'm currently working in a porting of this library to baremetal raspberry pi using circle framework, so this kind of component would be useful as i have a lot of processing power. here's the result so far: 
[image]

The next step would be to make this library work with the GPU DMA, so it would look realy fast and realtime. Professional embbed applications like car computers or embbeded PLC controllers would use it as your library is already better than solutions like ugfx. With just a little more widgets would best any commercial solution. The documentation is also better than any other gui library i've seen. The idea of implemeting the function of the driver in a simple struct is awesome, finally someone that is thinking of how to simplify the things in other hardware. Looking forward to the project, happy new year,

kisvegabor, 2017-12-25:
Hi, 
it's very good to see LittlevGL running on a Raspberry PI. 

You have mentioned a lot of interesting topics with great ideas and I'm very curious about them. May I ask you to create separate issues for them?
* Port to baremetal Raspberry PI
* Now widget ideas
* GUI designer

And the video **support**: 
It just came to my mind that LittlevGL draws the screen only when something happens. So if you place the buttons on the bottom and leave the rest of the screen then you can draw there (put the video) without any problem.

I'm waiting for the details of your project and the other things.

Happy new year!

kisvegabor, 2018-03-09:
I'm closing this issue because no activity happened for a few month. 
However we can reopen it any time or you can create a related issue! :)
Thank you!

closed 2018-03-09 by kisvegabor