lvgl/lvgl #49: usage of LV_INDEV_POINT_MARKER throws compiler errors

ramonschepers, 2017-12-08:
so i set the LV_INDEV_POINT_MARKER to 5 to see my presses(?) in the lvgl config file

buuuut.... my IDE throws compiling errors.

the complete error log:

```
C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c: In function 'indev_proc_point':

C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:286:9: error: unknown type name 'area_t'

         area_t area;

         ^

C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:287:13: error: request for member 'x1' in something not a structure or union

         area.x1 = (indev->act_point.x >> LV_ANTIALIAS) - (LV_INDEV_POINT_MARKER >> 1);

             ^

C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:288:13: error: request for member 'y1' in something not a structure or union

         area.y1 = (indev->act_point.y >> LV_ANTIALIAS) - (LV_INDEV_POINT_MARKER >> 1);

             ^

C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:289:13: error: request for member 'x2' in something not a structure or union

         area.x2 = (indev->act_point.x >> LV_ANTIALIAS) + ((LV_INDEV_POINT_MARKER >> 1) | 0x1);

             ^

C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:290:13: error: request for member 'y2' in something not a structure or union

         area.y2 = (indev->act_point.y >> LV_ANTIALIAS) + ((LV_INDEV_POINT_MARKER >> 1) | 0x1);

             ^

C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:291:18: warning: passing argument 1 of 'lv_rfill' from incompatible pointer type [-Wincompatible-pointer-types]

         lv_rfill(&area, NULL, LV_COLOR_MAKE(0xFF, 0, 0), LV_OPA_COVER);

                  ^

In file included from C:\Users\Ramon\Documents\Arduino\libraries\lvgl\src\lv_obj\lv_indev.c:16:0:

c:\users\ramon\documents\arduino\libraries\lvgl\src\lv_draw\lv_draw_rbasic.h:41:6: note: expected 'const lv_area_t * {aka const struct <anonymous> *}' but argument is of type 'int *'

 void lv_rfill(const lv_area_t * cords_p, const lv_area_t * mask_p,

      ^

```

kisvegabor, 2017-12-09:
Hi Ramon,
It is already fixed in dev-5.0 branch but if you update you need to do other changes as well (see dev-5.0 doc/README).
For now, I think it would be easier to fix the issue in your current version. You only need to  change `area_t area;` to `lv_area_t area;`

I hope that is all.

kisvegabor, 2017-12-09:
On Monday I can send you a .zip (like before) with the new version

kisvegabor, 2017-12-11:
Here is the new version. [lvgl_dev-5.0_arduino.zip](https://github.com/littlevgl/lvgl/files/1547000/lvgl_dev-5.0_arduino.zip)

Please follow the updates in your initialization. 
- Now only `disp_flush` is required (same as `map_fp`) but `lv_flush_ready();` need to be called at the end
-   `lv_tick_inc(1)` is used instead of `lv_tick_handler()`

```c
#include <lvgl.h>

void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{

  /*Write the coordinates like with disp map*/

  lv_flush_ready();
}

void hal_init(void)
{
    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    disp_drv.disp_flush = my_disp_flush;
    lv_disp_drv_register(&disp_drv);

    /*Initialize the graphics library's tick*/
    /*In a Timer call lv_tick_inc(1) in every milliseconds*/
}

void setup() {
  lv_init();

  hal_init();

  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino!");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

}

void loop() {
  // put your main code here, to run repeatedly:

  lv_task_handler();
  //delay_ms(3);
}
```

kisvegabor, 2017-12-21:
Fixed in v5.0.0

closed 2017-12-21 by kisvegabor