lvgl/lvgl #52: how to change input device event handling model from polling-event to post-event type

hotpeperoncino, 2017-12-25:
Hi. I want to change  input device event handling model from polling-event to post-event type.
I don't know whether the change has any advantage or not.
As for electric consumption, it looks that post-event model is better than polling-model.
is it possible for lvgl to prepare the framework for it ?

kisvegabor, 2017-12-26:
Hi,
lvgl will always poll via he input device `read` function, however in your read function you can use either polling or events.  For events I mean when an even (move, click etc) happens you can save what happened and and just return with it in the `read` function.

See  this as a reference: https://github.com/littlevgl/lv_drivers/blob/master/indev/mouse.c

hotpeperoncino, 2018-01-16:
Thank for your advice. I'll close this issue.

closed 2018-01-16 by hotpeperoncino