lvgl/lvgl #53: build on win pc_simulator err

helezh, 2017-12-26:
i using qt_mingw build err,and correct "int main(viod)"to "int winmain(int a,char *arg)" build ok and runing ok!

kisvegabor, 2017-12-26:
Hi,
Thanks!

It is specific to *qt_mingw* that you need to use *winmain* instead of *main* ?

helezh, 2017-12-26:
qt used cygwin gcc

kisvegabor, 2017-12-26:
And in other applications you also have to use winmain instead of main ?

helezh, 2017-12-27:
yes ,i pragraming pc-soft on win,the entry is winmain.now i have new problem,that is i used your lv-utils the    "fnt2c.py",but get many err.
"D:\Users\Administrator\Desktop\N3290x\lv_proj_pc_v4_2_0\lv_utils-master>python f
nt2c.py -f cjk -o cjko >ii
Traceback (most recent call last):
  File "fnt2c.py", line 444, in <module>
    main(sys.argv[1:])
  File "fnt2c.py", line 440, in main
    processConfig(conf)
  File "fnt2c.py", line 371, in processConfig
    source += makeFontSource(conf);
  File "fnt2c.py", line 326, in makeFontSource
    img, glyphs = loadFont(config)
  File "fnt2c.py", line 290, in loadFont
    font = minidom.parse(config.input_file)
  File "D:\Python27\lib\xml\dom\minidom.py", line 1918, in parse
    return expatbuilder.parse(file)
  File "D:\Python27\lib\xml\dom\expatbuilder.py", line 924, in parse
    result = builder.parseFile(fp)
  File "D:\Python27\lib\xml\dom\expatbuilder.py", line 207, in parseFile
    parser.Parse(buffer, 0)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 3, column 14
"

helezh, 2017-12-27:
i have sloved the problem,that is save as the *.fnt file to new *,fnt files(utf8 format, linux dont have the problem)

helezh, 2017-12-27:
using the fnt2c must install python27 pip pillow dom to support

helezh, 2017-12-27:
a new suggestion,many cjk fonts is NxN square,the "static const uint8_t cjko_width[] =  32,............." can not do this!

helezh, 2017-12-27:
here is chinese,example "你好！"

kisvegabor, 2017-12-27:
Hi,
thank you for the suggestions!

It means the max width (255) is not enough for CJK characters?

kisvegabor, 2017-12-27:
Or the width table is not required because all characters have the same width?

helezh, 2017-12-27:
yes,all the cjk character have the same height and width,the width table is not required! and all of cjk symbol large as 20,000! and the lv_utils can not cov cjk_0.png+cjk_1.png+cjk_2.png+cjk_3.png+cjk_4.png

helezh, 2017-12-27:
here is the files!

helezh, 2017-12-27:
cjk symbols unicode  0x4e00 to 0x9fbb is the base symbols

kisvegabor, 2017-12-27:
I see. You can increase the texture size in bmfont/export options to get 1 file.

kisvegabor, 2018-03-09:
As this issue already contains a lot of different topics I close it now.
If you have any new questions or ideas just open a new issue!
Thank you!

closed 2018-03-09 by kisvegabor