lvgl/lvgl #54: Has any low level interface in this LIB?

YipCharles, 2017-12-26:
LittlevGL is nice.
I figured it using some object drawing interface, but when I wana just draw a simple line or rectangle or just for testing the hardware, the Lib just stop in lv_refr_areas(), seems like I should enbale the area firstly?

kisvegabor, 2017-12-26:
Hi!

Maybe you know that there are two drawing methods: buffered and unbuffered. Learn more here: http://www.gl.littlev.hu/basics#drawing-and-rendering

In buffered mode (LV_VDB_SIZE > 0) you can't draw to the display directly because the buffering is controlled by the library.

In unbuffered mode (LV_VDB_SIZE == 0) you can draw to the display directly but your drawing might be "overdrawn"  when a refresh occurs.

I recommend rather to create and `lv_line` object to draw a line, or an `lv_obj` to create a simple rectangle.

In general lvgl not thinking in drawing (from the user side) but thinking in graphical objects. This way the library will know the graphical objects and can draw them correctly.

YipCharles, 2017-12-26:
That means I can create anything I want after create a object? (Such as an oscilloscope or a multi-meter.)

kisvegabor, 2017-12-26:
For an oscilloscope you can create a line object or a chart and set the point's values. See http://www.gl.littlev.hu/object-types

YipCharles, 2017-12-26:
Yep, I know that.
But if I want to redesign my oscilloscope, (will be more complex) , how?

kisvegabor, 2017-12-26:
If you mean you would like to add e.g. Volt/div or trigger mode just create drop down lists for them or for AC/DC coupling create button.

YipCharles, 2017-12-26:
Well, I mean there are lot of point and line, that I have to draw directly into the background, and the style just diff from LVGL, how about that?

kisvegabor, 2017-12-26:
I just don't ubdetstand why would you like to draw directly to the background. Please tell a little bit more details about your goals.

What kind of style do you need? Please tell it too.

YipCharles, 2017-12-27:
Well, for an example, I saw an Agilent Oscilloscope, I want to design like that. There are some dash line to  mark or measuring, some curve to display the FFT effect, such like those.

kisvegabor, 2017-12-27:
Well, actually dash line is not supported now but maybe it will in a next release. Or check http://gl.littlev.hu/services to request a new feature or use thin line instead.
But for FFT a chart can be used very well. And the whole GUI of an Agilent oscilloscope can be built from objects. It's much simpler then just drawing. E.g. just create a chart or button and the library will draw or redraw it when it is required. You just need to set the chart's data or a button's click action (a function to call when clicked).

YipCharles, 2017-12-27:
Ok. That will be more efficient.  : P
But what if I, whatsoever, want to have my own design object, (I just found "lv_objx_templ.c") Is that right to modify it?

kisvegabor, 2017-12-27:
Then you have two options:
1. Use lv_objx_templ but copy it like lv_myobj
2. Set a new design function for an object: lv_obj_set_design_func(line1, my_func)

Are you familiar with styles? You can modify the appearances with them.

kisvegabor, 2017-12-27:
Sorry, closed by mistake :)

YipCharles, 2017-12-27:
Fine. 
Well, that will be more expansible.

kisvegabor, 2017-12-27:
How do you mean?

I forget to say that I recommend the using of styles because they are created for users. The new object type and design function are more complicated and kind of hacks

YipCharles, 2017-12-27:
Yes, I know that will be better. I will prior use the objects existing.

BTW, what does "LV_MISC" means?

kisvegabor, 2017-12-27:
Lv_misc contains  not graphics but general stuffs like memory management,  task handling etc.

YipCharles, 2017-12-28:
It seems like a little bit heavy? I just undefined most of objects, anti-aliasing, shadow, but still cost a lot of time to compile and download the firmware. 
Do you have any idea?

YipCharles, 2017-12-28:
The flash size I will check latter, not small.

YipCharles, 2017-12-29:
For about 130K extra more with LVGL...

kisvegabor, 2017-12-29:
Have you disabled the unused fonts?

YipCharles, 2017-12-30:
I dis the used 40 Fonts, works, thanks.
BTW, it support UTF-8 encode, so does it mean I can display any language I want on the earth?

kisvegabor, 2017-12-30:
Basically yes. There are some "tricky" languages like Arabic or Thai which needs some things but e.g Chinease, Korean, Greek, Cyrillic etc is fully supported.

Read this to learn how the fonts work: 
http://www.gl.littlev.hu/basics#fonts

And this to see how ta add a new font:
http://www.gl.littlev.hu/font-converter

kisvegabor, 2018-01-19:
As a lot of question is answered here and we are far from the original question I close this issue.

closed 2018-01-19 by kisvegabor