lvgl/lvgl #55: Confuse of memory defining

YipCharles, 2017-12-30:
1. LV_MEM_SIZE
This is for LVGL working needed, there are lots of dynamic ram to be allocated.

2. LV_VDB_SIZE,  LV_VDB_ADR
I use RGB interface in my stm32f4, so this memory is my display area. The screen will change immediately as I change something in this area.

3. screen, LV_VDB2_ADR 
The same as VDB, another parellel memory for double buffering. When I operate VDB1, LCD controller will fetch from VDB2, and swap in next cycle.

Is that right?

kisvegabor, 2017-12-30:
1. LV_MEM_SIZE: correct. The library stores its sruffs here. (mainly the graphical object's data)
2. LV_VDB_SIZE: VDB is not your frame buffer. It's a graphics buffer of lvgl. It is used to make double buffered like graphics with only a few memory. Set to e.g. 10 kB. LV_VDB_ADR is an option to put the VDB into an external RAM. But it's better to keep it in the internal RAM so set LV_VDB_ADR to 0.
3. LV_VDB_DOUBLE uses 2 VDB for parallel execution of VDB flush to frame buffer and rendering to VDB. LV_VDB2_ADR is the same LV_VDB_ADR.

Learn more here about VDB: 
http://www.gl.littlev.hu/basics#drawing-and-rendering

Please look at the lv_boards repo. There is a project with the same MCU like yours.

Thanks for the feedback I will make the comments cearer in lv_conf.h

YipCharles, 2017-12-30:
Sorry, I made a mistake.
Look: 
typedef struct
{
    lv_area_t area;
    lv_color_t *buf;
}lv_vdb_t;
You mean VDB->buf NOT represents my frame buff?

YipCharles, 2017-12-30:
If NOT, why you still flush like this way?
lv_vdb_t * vdb_act = lv_vdb_get();
lv_disp_flush(vdb_act->area.x1, vdb_act->area.y1, vdb_act->area.x2, vdb_act->area.y2, vdb_act->buf);

kisvegabor, 2017-12-30:
No, it isn't. You have nothing to do with it just set its size. The library will handle it.

YipCharles, 2017-12-30:
Well, the driver: void (*disp_flush)(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p);

I should copy the color point data to my real frame buff, correct?

kisvegabor, 2017-12-30:
The flush function needs to copy the VDB to your framebuffer.

kisvegabor, 2017-12-30:
Yes, correct.

http://www.gl.littlev.hu/porting#display-interface

YipCharles, 2017-12-30:
Well, I had written a double frame buffering for the hardware so I confuse of VDB and VDB2, haha...
But now, it looks like a little difficult to copy VDB data to my display frame...

YipCharles, 2017-12-30:
Because you can't just update partial of the frame, then the other one will be blank...

kisvegabor, 2017-12-30:
LittlevGL is a complex library which does a lot of things instead of you. You don't need to write double buffering just copy VDB to the frame buffer's memory. To increase the performance increase the VDB size.

YipCharles, 2017-12-30:
LVGL did the job, but if I just copy VDB to frame, in the meanwhile LCD controller fetching from this frame, the screen will still be tearing, isn't it?

kisvegabor, 2017-12-30:
Yes, tearing can happen. In your flush function maybe you can trigger to the TFT refresh signal to synchronise.

kisvegabor, 2018-01-19:
I updated lv_conf_temp.h to be more descriptive in this field.

closed 2018-01-19 by kisvegabor