lvgl/lvgl #56: When clicking list item, focus or selection is gone.

gadget114514, 2017-12-30:
Hi. Thank you for your effort.

I'm trying to write a program like following.
My trouble is that mixing use mouse and keypad does not work well.
ind has "keyboard_read" as input function.

1. execute program, the list is opened.
2. "New" button has focus.
3. By using arrow key, we can change focus.
4. Clicking "New button", focus is gone.
5. The other click cannot recover focus of button.
6. There is no focus so using arrow key has no effect.

I'm using mono theme. I don't know it may be a cause of this behavior. 

void app_init(lv_indev_drv_t *ind)
{
    lv_theme_t *th = lv_theme_mono_init(210, NULL);

  /*Set the surent system theme*/
  lv_theme_set_current(th);

  lv_group_t *group = lv_group_create();
  /*Crate the list*/
  lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
  lv_obj_set_size(list1, 130, 170);
  lv_obj_align(list1, NULL, LV_ALIGN_IN_TOP_LEFT, 20, 40);
  lv_list_set_sb_mode(list1, LV_SB_MODE_AUTO);

  /*Add list elements*/
  lv_list_add(list1, SYMBOL_FILE, "New", list_release_action);
  lv_list_add(list1, SYMBOL_DIRECTORY, "Open", list_release_action);
  lv_list_add(list1, SYMBOL_CLOSE, "Delete", list_release_action);
  lv_list_add(list1, SYMBOL_EDIT, "Edit", list_release_action);
  lv_list_add(list1, SYMBOL_SAVE, "Save", list_release_action);

  lv_group_add_obj(group, list1);
  lv_group_focus_obj(list1);
  lv_indev_set_group(ind, group);
}

kisvegabor, 2017-12-31:
Hi,

thank you for telling your experience!

You are right, it wasn't worked well. I fixed it in the `beta` branch of `lvgl`.

Does it work properly now?

gadget114514, 2018-01-16:
Thanks, it works well.

kisvegabor, 2018-01-16:
Great!
Your welcome!

closed 2018-01-16 by kisvegabor