lvgl/lvgl #57: Is there any checkbox objects?

YipCharles, 2018-01-02:
And, can we just change the button object item's background to a image?

kisvegabor, 2018-01-02:
Yes, it is. You find all the object types here: https://littlevgl.com/object-types

There is no option to set an image as button's background. You can create a button and an image on it. 
It looks a useful feature! I will add this option in the next release.

YipCharles, 2018-01-16:
Yep, I mean, drawing a button by some lines and curves was good, but some times putting 2 pic on it (another for pressed status) will be easier and redesignable, making it possible to change the button looking on different modes.

kisvegabor, 2018-01-16:
I will add this feature in the next release. A good name could be `lv_icon`. It is a clickable image with 2 image sources: one for released one for pressed

kisvegabor, 2018-03-09:
The idea of lv_icon  was also mentioned in #144.
It will added in an upcoming release.

closed 2018-03-09 by kisvegabor