lvgl/lvgl #58: This GUI library is very cool, I hope you can continue to update

jason-sophia, 2018-01-03:
thank you very much for sharing your code

kisvegabor, 2018-01-04:
Hi,
thank you very much. I'm glad to hear that you like it!

Maybe you know there is a Donation opportunity on https://littlevgl.com 
Even a few dollars could help to buy new dev boards to test and extend the library and bring new feature for you!

jason-sophia, 2018-01-08:
OK，I will certainly make a donation, but first I have to apply for a visa or Matercard

kisvegabor, 2018-01-08:
Thank you in advance!

You just need a PayPal account. If you don't have an appropriate credit card  a friend or family member can send money there. 
Having a PayPal account is cool thing anyway to order from Ebay, Amazon etc.

jason-sophia, 2018-01-08:
OK

jason-sophia, 2018-01-12:
@kisvegabor I have made a donation, have you got it?

kisvegabor, 2018-01-12:
Hi,
I got a Donation from HO KIT MING. Was it you?

jason-sophia, 2018-01-15:
@kisvegabor Yes, she is my friend

kisvegabor, 2018-01-15:
@jason-sophia Thank you very mush then! I also wrote an e-mail to her to say thank you!
Now you are listed here: https://littlevgl.com/donors
If you want I can change the name.

kisvegabor, 2018-03-12:
I close this thread to keep the list of issues clear!

Thank you very much for the support!

closed 2018-03-12 by kisvegabor