lvgl/lvgl #59: Support for line based memory LCDs?

iromero91, 2018-01-03:
I am in the process of writing a driver for SHARP memory LCDs like the [LS027B7DH01](http://www.mouser.com/ds/2/365/LS027B7DH01-542063.pdf) These are either monochrome or 3 bit RGB displays that have built in memory. One limitation of these displays is that they can only be updated a WHOLE line at a time, and there is no way to read back the current display contents. As it stands i'd have to keep a buffer of the whole screen in memory to implement the driver (which I don't want, things barely fit as is). 

I have not looked too hard yet on how the rendering works in this library, but i was wondering if it would be possible to guarantee that the calls to my_disp_flush() would be whole lines at a time?

qwert1213131, 2018-01-04:
LS013B7DH06 is 3 bit RGB ,128x128

kisvegabor, 2018-01-04:
Hi,
right now there is no direct option to force the library to always refresh a whole line. But you can do this very simply in your forked version:
In **lv_refr.c** the `lv_inv_area()` function makes an area invalid and asks the library to refresh it. Your only task is to modify the area to always be screen wide. 

Insert:
```c
com_area.x1 = 0;
com_area.x2 = LV_HOR_RES-1;
```
before this line: https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_refr.c#L118

iromero91, 2018-01-08:
Thanks! That modification worked. I read in previous issues that this library expects the driver to handle screen rotation, for a 90 degree rotated screen i should modify y1 and y2 instead, right?

Do you think you could merge patches for this? perhaps with a couple of variables like: LV_VDB_FLUSH_WHOLE_ROW, LV_VFB_FLUSH_WHOLE_COLUMN

iromero91, 2018-01-08:
I just had a thought that maybe the VDB cant work in columns instead of rows, if so, do you think that can be changed so the screen rotation in the driver doesn't have to store the whole screen in a buffer?

kisvegabor, 2018-01-09:
Great!

The area in _disp_fush_ can be any type of rectangle: 1 row, 1 col or "normal". If you would like to change to orientation from landscape to portrait you need change LV_HOR_RES and LV_VER_RES (e.g  LV_HOR_RES 240 and LV_VER_RES 320).
And in your special case now a line in your display controller become a column in LittlevGL. So apply the above described modification for *y* coordinates and LV_VER_RES:
```c
com_area.y1 = 0;
com_area.y2 = LV_VER_RES-1;
```

A define for it seems absolutely reasonable! If you want to contribute you can add this update to `dev-5.1` branch. Please call the new defines to LV_INV_FULL_ROW and LV_INV_FULL_COL and don't forget to add them to *lv_conf_templ.h* too. Below `#define LV_INV_FIFO_SIZE` seems a good place for it.

iromero91, 2018-01-09:
Will do, expect a pull request soon

closed 2018-01-09 by iromero91