lvgl/lvgl #62: problems porting to STM32L496

jason-sophia, 2018-01-08:
I did everything in the porting guide, the The program still can not work.
the display driver is OK, could you help me ?
the code is in attach.
[gui_lvgl.zip](https://github.com/littlevgl/lvgl/files/1611054/gui_lvgl.zip)

kisvegabor, 2018-01-08:
Hi!

There are 2 things:
1. I noticed that I made a mistake in lv_tutorial_porting (already fixed): the x and y coordinates were mixed. 
2. If you are using 32 bit colors with alpha then you should set the alpha to cover.

So the disp_flush function:

 ```c
void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
	int32_t x = 0;
	int32_t y = 0;
	uint32_t color_mod;

	for (y = y1; y <= y2; y++)                          /*Pick the next row*/
	{
		for (x = x1; x <= x2; x++)                     /*Pick the a pixel in the row*/
		{
			color_mod = color_p->full | 0xFF000000;         /*Set alpha to cover*/
			BSP_LCD_DrawPixel(x,y, color_mod);
			color_p++;
		}
	}

	lv_flush_ready();
}
```

I will update the library to pre-fill the upper  8 bit of 32 (and 24) bit color with 0xFF or given value.

Is it working now?

jason-sophia, 2018-01-08:
It works! 
Thank you very much !

kisvegabor, 2018-01-08:
Your welcome!

closed 2018-01-08 by kisvegabor