lvgl/lvgl #63: Colors mixed up on big-endian architectures (bitfields + unions)

shoggoth77, 2018-01-08:
Types lv_colorNN_t conveniently uses C bitfields to access the individual red, green and blue components of a pixel. As some clever person (who?) once said - "as a programming language, C provides lots of rope two hang yourself with", and in this context, this means colours get mixed up on big endian machines. Why? C does not define the order of bitfield-members. And indeed, the order differs between different compilers and different endians.

I've verified this empirically. A workaround which seems to work is to switch the order of the red and blue components in LV_COLOR_MAKE based on the current endian (lv_color.h):

`#if __BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__
#if LV_COLOR_DEPTH == 1
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){(b8 >> 7 | g8 >> 7 | r8 >> 7)})
#elif LV_COLOR_DEPTH == 8
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{b8 >> 6, g8 >> 5, r8 >> 5}})
#elif LV_COLOR_DEPTH == 16
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{b8 >> 3, g8 >> 2, r8 >> 3}})
#elif LV_COLOR_DEPTH == 24
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{b8, g8, r8}})
#endif
#else
#if LV_COLOR_DEPTH == 1
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){(r8 >> 7 | g8 >> 7 | b8 >> 7)})
#elif LV_COLOR_DEPTH == 8
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{r8 >> 6, g8 >> 5, b8 >> 5}})
#elif LV_COLOR_DEPTH == 16
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{r8 >> 3, g8 >> 2, b8 >> 3}})
#elif LV_COLOR_DEPTH == 24
#define LV_COLOR_MAKE(r8, g8, b8) ((lv_color_t){{r8, g8, b8}})
#endif
#endif`

Note however that I've only verified that this works on 16-bit colour depths.

kisvegabor, 2018-01-08:
Hi,

I see. 
I'm already planning a config feature like:` LV_COLOR_BGR` to change the color fields order.
Calling this feature to BGR seems a little bit more intuitive to me. What do you think?

shoggoth77, 2018-01-08:
Hi :)

I'd say yes - and - no. This might lead the user to associate this with the actual pixel format, which is not necessarily true (e.g. in this case, it's r5g6b5 both on my big-endian architecture and on the little-endian one - being forced to select BGR to make the colours appear right feels a bit counter-intuitive). 

Basically there is probably no "one size fits all"-solution unless one skips the use of bitfields for this altogether (some exotic compiler could in theory change the order to GBR or BRG anyway, due to the lack of a standardised bitfield order in C). On the other hand, ditching bitfields may have a negative impact e.g. on compiler optimisations, so perhaps the current design is a good tradeoff. In such case, it could make more sense to document the issue rather than changing the code.

kisvegabor, 2018-01-08:
You are right, it could mislead the users about the issue. 
I will update the code according to your suggestion!

kisvegabor, 2018-01-19:
I added your code in v5.0.2.

Thank you for that!

closed 2018-01-19 by kisvegabor