lvgl/lvgl #64: Binary literals is a GCC extension

shoggoth77, 2018-01-08:
Hi,

Again, a very minor issue, please stop me if I'm polluting the issue tracker.

Binary literals is a GCC-extension and should probably be replaced with hexadecimal counterpart, to ensure compatibility with exotic compilers (this is, after all, a library for embedded systems - exotic compilers is norm rather than exception).

e.g. 0b10101010 -> 0xAA. There are not that many occurrences, but anyway.

kisvegabor, 2018-01-08:
Hi,

I'm very happy if I get feedback and tips how to improve the library. That why I shared it. So all of your suggestions are very welcome!

I never heard that using 0b.... representation could be a problem. But if it is, I will change them to 0x...

Than you for the suggestion!

shoggoth77, 2018-01-08:
Well, it is commonly supported, but afaik it's an extension rather than part of the C language. Wether or not it is a problem - well obviously it isn't, because it's mega-easy to port this library to just about any platform. I'm nitpicking a bit.

kisvegabor, 2018-01-08:
And it's also mega-easy to solve that potential issue! :)

kisvegabor, 2018-01-19:
I changed 0b... defination to 0x... as you suggested. They are in v5.0.2
Thank you for the idea!

closed 2018-01-19 by kisvegabor