lvgl/lvgl #65: Endian issue - images (const data in demo.c)

shoggoth77, 2018-01-08:
When testing the demo.c I noticed that the background isn't drawn correctly on big endian architectures.

This is easily fixed if the image header is declared using the lv_img_raw_header_t struct rather than binary data.

Example:
```
const struct
{
  lv_img_raw_header_t header;
  lv_color_int_t data[];
} img_bubble_pattern =
{
/*HEADER
   Width = 235
   Height = 480
   Transp: 0
   Color depth: 16*/
  { 235, 480, 0, 2, 0 },
  /*IMAGE DATA*/
  {
     19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479,
     19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479,
     19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479, 19479
     
     ...
     
     10866,
  }
};
```

This works out of the box on big- and little endian machines. I suspect this to cure other potential issues as well (struct members are aligned differently on different compilers, which means hard-coded binary data for the header can fail badly on some architectures).

... I guess this is generated code, so it could probably make sense to adjust the code generator.

shoggoth77, 2018-01-08:
... oops - this means C99 is required, because of lv_color_int_t data[] (variable length struct members is not standard prior to C99).

kisvegabor, 2018-01-08:
There are other Variable Length Arrays in the code to consume only as much memory as really needed. So applying it here is not problem.

The only issue is that it breaks backward compatibility so a new major release will be required. And I released v5 in last month... :D

shoggoth77, 2018-01-08:
It's no biggie, I'd just like to record my findings somewhere in case it's useful for someone :)

(btw - v5 is a great improvement, hats off for great work on that one)

kisvegabor, 2018-01-08:
Keep this great work! I will try to solve what I can!

(Thank you! :) )

kisvegabor, 2018-07-15:
In v5.1 the images are generated in a byte array instead of a  color array which solves the Big endian issue.

Thank you for pointing this out!

closed 2018-07-15 by kisvegabor