lvgl/lvgl #67: compilation failure if LV_INDEV_READ_PERIOD = 0

hreichert, 2018-01-09:
Defined in `lv_conf.h`:
```
/*Input device settings*/
#define LV_INDEV_READ_PERIOD 0
```

Compilation error:
> [Error] lv_indev.c@60,25: too few arguments to function 'lv_task_create'                                                                          
> [ERROR] ./libs/littlevgl/lvgl/lv_core/lv_indev.c: In function 'lv_indev_init':                                                                    
> ./libs/littlevgl/lvgl/lv_core/lv_indev.c:60:25: error: too few arguments to function 'lv_task_create'

Temporary fix to make it work locally:
(added NULL as fourth argument)
```
diff --git a/lv_core/lv_indev.c b/lv_core/lv_indev.c
index 25a4157..c180478 100644
--- a/lv_core/lv_indev.c
+++ b/lv_core/lv_indev.c
@@ -57,7 +57,7 @@ void lv_indev_init(void)
 #if LV_INDEV_READ_PERIOD != 0
     indev_proc_task_p = lv_task_create(indev_proc_task, LV_INDEV_READ_PERIOD, LV_TASK_PRIO_MID, NULL);
 #else
-    indev_proc_task_p = lv_task_create(indev_proc_task, 1, LV_TASK_PRIO_OFF); /*Not use lv_indev_proc*/
+    indev_proc_task_p = lv_task_create(indev_proc_task, 1, LV_TASK_PRIO_OFF, NULL); /*Not use lv_indev_proc*/
 #endif
 
     lv_indev_reset(NULL);   /*Reset all input devices*/
```

kisvegabor, 2018-01-10:
Thank you!
I fixed in the `beta` branch. Now `LV_INDEV_READ_PERIOD 0` also disables a lot of functions in lv_indev.c to make the code smaller with this configuration.

hreichert, 2018-01-10:
Thank you very much! Really appreciate your efforts.

closed 2018-01-10 by hreichert