lvgl/lvgl #69: how to define the  LV_DPI value?

qwert1213131, 2018-01-10:


kisvegabor, 2018-01-10:
Hi,

In this case you have 240 pixels in 1.54". So you have got 240/1.54 pixels in 1" -> LV_DPI 155

Keep in mind the purpose of LV_DPI is to set the initial sizes and spaces to the same on every screen. 
For example a default width (for something) can be LV_DPI to make 1 inch wide on every screen.
However if you want smaller objects you can decrease LV_DPI to cheat LittlevGL.

qwert1213131, 2018-01-10:
hi,
Thank you so much, and I finally got it.
I'm studying version 5.
If you can share some of the wearable UI designs or other design routines, you can attract more people to join.
This project is really great.

kisvegabor, 2018-01-10:
Great! Thank you, glad to hear that you like it! :)

Do you mean share some examples for wearable UI?

qwert1213131, 2018-01-11:
Yes, share some wearable design examples and share more video.
I'm getting started, and I'll share some video or pictures if I'm going to make a progress.
Thank you！

kisvegabor, 2018-01-11:
Great, thank you in advance!

The offer on my hompeage is real: if you have a LittlevGL based GUI which looks impressive I kindly put it to my homepage!

qwert1213131, 2018-01-11:
I'm going to use this GUI actively.
It will be shared at the same time.
Thank you very much.
I've been trying to learn some examples today, and this hardware button mapping is very useful.
[image]

kisvegabor, 2018-01-11:
Oh, looks nice! :) Keep this great work!
I'm waiting for the updates!

kisvegabor, 2018-01-19:
As the question is answered I close this issue.

closed 2018-01-19 by kisvegabor