lvgl/lvgl #70: button style looks incorrect in mono theme

hotpeperoncino, 2018-01-13:
in mono theme, button looks incorrect in selected state.
The color of text of label set in button is not changed to white when the button is selected.

  lv_obj_t *btn1 = lv_btn_create(scr, NULL);
  /*Add a label to the button*/
  lv_obj_t *label = lv_label_create(btn1, NULL);
  lv_label_set_text(label, "Normal");
  lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, back_to_main_button_onclick);
...
...
  lv_group_focus_obj(btn1);

hotpeperoncino, 2018-01-13:
The mono theme is set by following fragment.
   th = lv_theme_mono_init(210, NULL);
  lv_theme_set_current(th);

kisvegabor, 2018-01-13:
Hi,
I validated it. It was a tricky one:
1. There was a bug in LittlevGL about inheriting focused object's styles by a label. Please pull `lvgl/beta`
2. The default style modifier function (which updates the style of the focused object) is designed for colorful themes. So need to write your own. It's very easy:

```c
/*New style modifier function*/
static void my_group_style_mod(lv_style_t *style)
{
   /* Modify style properties (colors, width etc.) 
    * but only those which don't modify the object size. 
    * So padding, font, letter_space etc. are forbidden. */
    style->body.border.width = 4;
    style->body.border.color = LV_COLOR_BLACK;
}

...

lv_group_t *g = lv_group_create();
lv_group_set_style_mod_cb(g, my_group_style_mod);
```

To avoid this issue in the future I will add different default style_mod function for `LV_COLOR_DEPTH   1`

hotpeperoncino, 2018-01-16:
Hi. Thank you for your response. I'll check it.

kisvegabor, 2018-01-19:
Hi,
have you tried is?

kisvegabor, 2018-01-23:
According to conversation in other issues I suppose it already solved.

closed 2018-01-23 by kisvegabor