lvgl/lvgl #71: [Styling guide] clang-format and git commit hooks

C47D, 2018-01-13:
Hi,

Thanks for sharing your library, i haven't tested it yet but i will :).

I was reading the CONTRIBUTING guide [1] and thought that it may be a good idea to use clang format and a git commit hook to automate the code style formatting (just in case the contributor forget about formatting the code according to your style guide), clang-format it's not perfect but it can help.

The contributors will need to have clang format (or clang) installed and a .clang-format file would be needed on the root of this repo to specify the format style.

I summit this idea, it may be good if you and the regular contributors discuss about it.

[1] https://github.com/littlevgl/lvgl/blob/master/docs/CONTRIBUTING.md#styling-guide

kisvegabor, 2018-01-15:
Hi,

Thank you for the idea. It means the code is automatically formatted when somebody commits? If yes, then it sound great!
 
Unfortunately I never used git commit hooks, so to add this features I need to learn them.
If you are familiar with them and can do it with a small effort, I would highly appreciate any help in this field.

kisvegabor, 2018-03-12:
I close this issue because there was no activity here in the past few month. 
If somebody would like to contribute in this field feel free to re-open the issue.

C47D, 2018-04-03:
This is Work In Progress.
Hi, sorry for the loooong time it took me to reply, i'm using clang-format 6, this is the .clang-file (you can test it and improve it):

> Language:        Cpp
> AccessModifierOffset: -2
> AlignAfterOpenBracket: Align
> AlignConsecutiveAssignments: false
> AlignConsecutiveDeclarations: false
> AlignEscapedNewlines: Right
> AlignOperands:   true
> AlignTrailingComments: true
> AllowAllParametersOfDeclarationOnNextLine: true
> AllowShortBlocksOnASingleLine: true
> AllowShortCaseLabelsOnASingleLine: false
> AllowShortFunctionsOnASingleLine: All
> AllowShortIfStatementsOnASingleLine: true
> AllowShortLoopsOnASingleLine: false
> AlwaysBreakAfterDefinitionReturnType: None
> AlwaysBreakAfterReturnType: None
> AlwaysBreakBeforeMultilineStrings: false
> AlwaysBreakTemplateDeclarations: false
> BinPackArguments: true
> BinPackParameters: true
> BraceWrapping:   
>   AfterClass:      false
>   AfterControlStatement: false
>   AfterEnum:       false
>   AfterFunction:   false
>   AfterNamespace:  false
>   AfterObjCDeclaration: false
>   AfterStruct:     false
>   AfterUnion:      false
>   AfterExternBlock: false
>   BeforeCatch:     false
>   BeforeElse:      false
>   IndentBraces:    false
>   SplitEmptyFunction: true
>   SplitEmptyRecord: true
>   SplitEmptyNamespace: true
> BreakBeforeBinaryOperators: None
> BreakBeforeBraces: Custom
> BreakBeforeInheritanceComma: false
> BreakBeforeTernaryOperators: true
> BreakConstructorInitializersBeforeComma: false
> BreakConstructorInitializers: BeforeColon
> BreakStringLiterals: true
> ColumnLimit:     80
> CommentPragmas:  '^ IWYU pragma:'
> CompactNamespaces: false
> ConstructorInitializerAllOnOneLineOrOnePerLine: false
> ConstructorInitializerIndentWidth: 4
> ContinuationIndentWidth: 4
> Cpp11BracedListStyle: true
> DerivePointerAlignment: false
> DisableFormat:   false
> ExperimentalAutoDetectBinPacking: false
> FixNamespaceComments: true
> ForEachMacros:   
>   - foreach
>   - Q_FOREACH
>   - BOOST_FOREACH
> IncludeBlocks:   Preserve
> IncludeCategories: 
>   - Regex:           '^"(llvm|llvm-c|clang|clang-c)/'
>     Priority:        2
>   - Regex:           '^(<|"(gtest|gmock|isl|json)/)'
>     Priority:        3
>   - Regex:           '.*'
>     Priority:        1
> IncludeIsMainRegex: '(Test)?$'
> IndentCaseLabels: false
> IndentPPDirectives: None
> IndentWidth:     2
> IndentWrappedFunctionNames: false
> JavaScriptQuotes: Leave
> JavaScriptWrapImports: true
> KeepEmptyLinesAtTheStartOfBlocks: true
> MacroBlockBegin: ''
> MacroBlockEnd:   ''
> MaxEmptyLinesToKeep: 1
> NamespaceIndentation: None
> ObjCBlockIndentWidth: 2
> ObjCSpaceAfterProperty: false
> ObjCSpaceBeforeProtocolList: true
> PenaltyBreakAssignment: 2
> PenaltyBreakBeforeFirstCallParameter: 19
> PenaltyBreakComment: 300
> PenaltyBreakFirstLessLess: 120
> PenaltyBreakString: 1000
> PenaltyExcessCharacter: 1000000
> PenaltyReturnTypeOnItsOwnLine: 60
> PointerAlignment: Middle
> RawStringFormats: 
>   - Delimiter:       pb
>     Language:        TextProto
>     BasedOnStyle:    google
> ReflowComments:  true
> SortIncludes:    true
> SortUsingDeclarations: true
> SpaceAfterCStyleCast: false
> SpaceAfterTemplateKeyword: true
> SpaceBeforeAssignmentOperators: true
> SpaceBeforeParens: ControlStatements
> SpaceInEmptyParentheses: false
> SpacesBeforeTrailingComments: 1
> SpacesInAngles:  false
> SpacesInContainerLiterals: true
> SpacesInCStyleCastParentheses: false
> SpacesInParentheses: false
> SpacesInSquareBrackets: false
> Standard:        Cpp11
> TabWidth:        4
> UseTab:          Never
> ...

All that options are described here: [ClangFormat Style Options](https://clang.llvm.org/docs/ClangFormatStyleOptions.html).

I have checked and this can help you with the code formatting (just the style), it can't change the constants to uppercase and so on. One more thing to do is integrate this with git (there's a small description about this at the end of the link above).

[clang-format.zip](https://github.com/littlevgl/lvgl/files/1870169/clang-format.zip)

kisvegabor, 2018-04-03:
Thank you very much!
I will check it soon and get beck to you! :)

C47D, 2018-04-03:
The set of options of the .clang-format file i attached can be reduced to just use the options of your interest, i already removed some options that works for Java source files.

For adding the pre-commit hook you can see how it's done [here](https://github.com/arduino/ArduinoCore-arc32/wiki/Creating-a-pre-commit-hook-for-clang-format).

I suggest you to test it on a separate file and not directly on the repo source code, at least until everything is tested and working as you want.

Carlos

PD: I would like to control an OLED display (bicolor, via I2C) with lvgl, i already saw that the code on the drivers repo is getting reformatted so, do you know of any tutorial using lvgl and monochrome displays?

kisvegabor, 2018-04-03:
Thank you very much Carlos! It seems super easy to set up!
I've no experience with hooks ( yet :) ) and I'm wondering everybody should do these setup steps our it will just work for them?

kisvegabor, 2018-04-03:
Oh I just found it myself in the hook script: it checks weather clag-format is installed or not. So I will update the CONTRIBUTING.md with a suggestion to install clang-format.

C47D, 2018-05-27:
Hi,

Sorry for the late reply i didn't got any notifications from github. I will take a look at the problems you're getting and try to reply back, i had always used clang-format from the terminal.

kisvegabor, 2018-05-29:
Hi,

No problem :)

I used it from a Linux terminal as well.

C47D, 2018-06-12:
Hi,

I also used it from the Linux and Windows terminal, i will try to take a look at it (the hooks).

kisvegabor, 2018-07-15:
Hi,

Meanwhile, `astyle` support was added by @aenchevich . As it does code formatting I close this because the original goal is reached.

Thank you for the controbution!

closed 2018-07-15 by kisvegabor